qmk/qmk_firmware #18900: [Bug] VIA not compatible with keyboards flashed with custom firmware outisde this repo

TheBosoYolo, 2022-10-28:
### Describe the Bug

Keyboards like keychron k8 pro (https://github.com/keychron/qmk_firmware/tree/bluetooth_playground) flashed with custom firmware do not get recognized by VIA even with VIA_ENABLE: YES or something in rules.mk

### Keyboard Used

keychron/k8_pro

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-k8-pro-qmk-via-wireless-mechanical-keyboard

### Operating System

Windows

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: C:/Users/BosoYolo/Desktop/qmk_firmware
Ψ Detected Windows 10 (10.0.22000).
Ψ Git branch: master
Ψ Repo version: 0.18.14
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2022-10-17 09:34:59 +0800 (3288e3fcf8) -- Merge branch 'qmk:master' into master
Ψ - Latest upstream/master: 2022-10-28 18:48:08 +0800 (6514c25ca6) -- Docs: fix a typo (#18896)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2022-10-17 07:30:49 +1100 (f731e2789f) -- Allow Fedora to update ChibiOS configs. (#18698)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-06-11 09:09:45 +0000 --  (f836d24b0)
Ψ - lib/chibios-contrib: 2022-07-27 10:46:15 +0200 --  (d03aa9cc)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-05-17 19:19:01 +0200 --  (07edde8)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

other keyboards not located directly in qmk/qmk_firmware repo

### Additional Context

qmk doctor output is not relevant/not essencial information

tzarc, 2022-10-28:
Not a QMK issue -- VIA requires keyboards to be in mainline QMK.

closed 2022-10-28 by tzarc

thompson-vii, 2022-10-30:
sideload the json file from the product page