qmk/qmk_firmware #18928: [Bug] Satisfaction75 hangs hard every few seconds

DarkMio, 2022-11-01:
### Describe the Bug

When building from current master the satisfaction hangs often and repeatedly. It hangs on system, the OLED stops refreshing (which keypress indicators) and HID reports stay stuck on keypress down.

This issue has been verified by @tzarc yesterday.

A build from Feb 19 (sorry, I do not have the revision anymore) works fine.

```
[qmk_firmware]$ history | grep sat
   56  qmk flash -kb cannonkeys/satisfaction75/rev1 -km via
  138  history | grep sat
[qmk_firmware]$ git branch
* master
[qmk_firmware]$ git log -1
commit 652b10e6a76c07f59dd3199b7d167fc65722a98c (HEAD -> master, upstream/master, origin/master, origin/HEAD)
Author: Anton Chernenko <ch3rny777@gmail.com>
Date:   Mon Oct 31 03:57:23 2022 +0200

    [Keyboard] Add Hubble keyboard (#18718)

    Co-authored-by: Anton Chernenko <anton.chernenko@chaintrust.it>
```

### Keyboard Used

cannonkeys/satisfaction75

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

shsieh, 2022-11-21:
I am having the exact same issue. When I revert the commit to early March, the problem is gone.

There are no changes in the S75 folder. So the cause is definitely somewhere else. The unresponsive issue seems to due to some sort of blocking.

drashna, 2022-12-10:
If you disable VIA, encoder and/or oled, does this continue to happen?

ThunderrGodd, 2023-06-05:
> If you disable VIA, encoder and/or oled, does this continue to happen?

I gave it a try with non-VIA keymap, and it seems the bug not gonna happen.
Still works fine with combo and tap-dance keycode in the keymap.

awkannan, 2023-06-19:
I can no longer repro this on QMK master using a clean build. Can anyone else?