qmk/qmk_firmware #18931: Plans to solve the RDP issue?

LukasKubicek, 2022-11-01:
Hello good people. I have to work in RDP sessions almost every day. Mostly Windows 10 to remote Windows 10.

When in RDP, my keyboard (Moonlander configured with Oryx) occasionally produces wrong characters, it seems to be nondeterministic, but it is enough to make it unusable for work. On the local machine everything works correctly.

It seems that it is known issue for many years:
https://github.com/qmk/qmk_firmware/issues/1037

But there are only some non ideal workarounds.
Is there some plan to fix this RDP issue? Did someone maybe try to contact developers of RDP and consult with them? Or is all hope lost? In today's world of remote work I imagine this must be a troubling issue for many people with QMK kyeboards.

tzarc, 2022-11-01:
The plan is to accept PRs from people who have the issue and can provide a solution.
I too spend the majority of my day in RDP sessions, and have no such problem, I'm afraid.

JayceFayne, 2022-11-28:
Probably related to https://docs.qmk.fm/#/mod_tap?id=caveats / https://docs.qmk.fm/#/one_shot_keys relevant section:
> You may also run into issues when using Remote Desktop Connection on Windows. Because these keycodes send key events faster than a human, Remote Desktop could miss them. To fix this, open Remote Desktop Connection, click on “Show Options”, open the “Local Resources” tab, and in the keyboard section, change the drop down to “On this Computer”. This will fix the issue, and allow the characters to work correctly.

I never had any issues with RDP sessions but I don't use mod-tap or one shot keys.

Lingxi-Li, 2022-12-07:
Simple key-to-key remap seems to work reliably for me in full-screen RDP, but not macros (missing modifier strokes). Now I know that user mode key remap does not work reliably with RDP due to reasons discussed [here](https://github.com/microsoft/PowerToys/issues/3614). Wonder whether this applies to QMK macros. That is, while key-to-key remap works at firmware level, but macros work at user mode and hence the issue?

drashna, 2022-12-10:
@LukasKubicek Is this only for shifted/modded keycodes? 

If so, does adding a `wait_ms(TAP_CODE_DELAY)` before and/or after this line help:
https://github.com/qmk/qmk_firmware/blob/89fdafc27d2dc8652886950655352ce8628a5d57/quantum/action.c#L386

kayjtea, 2022-12-17:
A minor tweak to 

https://github.com/qmk/qmk_firmware/issues/1037#issuecomment-276106147

fixed issues with auto shift and with sym layer not always registering the correct key for me.  I use RDC to connect to locally running Linux VM. The commit is here: 

https://github.com/kayjtea/miryoku_qmk/commit/04777b6ea679e862e1a7ac2e66c94263dd9be76b

closed 2023-04-18 by github-actions[bot]