qmk/qmk_firmware #18970: [Bug] #18643 Appears to have regressed automatic mouse layer

chrishoage, 2022-11-05:
### Describe the Bug

When using `AUTO_MOUSE_DEFAULT_LAYER` where the mouse layer is the same layer as the keymap of `MO(MOUSE_LAYER)` (e.g. `AUTO_MOUSE_DEFAULT_LAYER 1` and `MO(1)` the layer is immediately lowered. 

I noticed I would receive two `mousekey` logs in qmk console 

```
QMK:Spleeb:1: mousekey [btn|x y v h](rep/acl): [00|0 0 0 0](0/0)
QMK:Spleeb:1: mousekey [btn|x y v h](rep/acl): [00|0 0 0 0](0/0)
```

Where as on the last good commit I would only get one `mousekey` log when triggering `MO(1)

I bisected back to https://github.com/qmk/qmk_firmware/commit/a69ab05dd687cb9aa38e0c125e4f64956c7da6c7 and [elep](https://github.com/qmk/qmk_firmware/commit/594dcbf66c5ddc9865bd0243f8d4215e8c034707) was the last good commit. 

To reproduce you seem to just need to have a MO(1) and `set_auto_mouse_layer(1);` / `AUTO_MOUSE_DEFAULT_LAYER 1`. I rebased my work on develop and noticed this regression - I then bisected back to the working commit. I made no other changes to my code 

### Keyboard Used

N/A (working on firmware for a new keyboard and found this issue) 

### Operating System

Linux

### qmk doctor Output

```text
~/projects/qmk_firmware hoage/sctratchpad* 12s
❯ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /home/chris/projects/qmk_firmware
Ψ Detected Linux.
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32-dfu' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'md-boot' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Git branch: hoage/sctratchpad
Ψ Repo version: 0.18.15
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest hoage/sctratchpad: 2022-11-05 17:05:19 +0000 (69fa02c37d) -- Align audio keycode names (#18962)
Ψ - Latest upstream/master: 2022-11-05 10:55:42 -0700 (e3a97eb1c8) -- [Keyboard] Add Work Louder Numpad (#18912)
Ψ - Latest upstream/develop: 2022-11-05 17:05:19 +0000 (69fa02c37d) -- Align audio keycode names (#18962)
Ψ - Common ancestor with upstream/master: 2022-10-31 03:57:23 +0200 (652b10e6a7) -- [Keyboard] Add Hubble keyboard (#18718)
Ψ - Common ancestor with upstream/develop: 2022-11-05 17:05:19 +0000 (69fa02c37d) -- Align audio keycode names (#18962)
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-09-18 10:01:17 +0000 --  (0e9d558b5)
Ψ - lib/chibios-contrib: 2022-10-03 18:09:41 +0200 --  (bb8356fb)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

I am happy to provide my unfinished keyboard code if it is helpful to reproduce this issue.

tzarc, 2022-11-06:
@chrishoage would you mind updating `develop`?
#18977 was merged in an attempt to fix it, would be good to get confirmation from your end that it is indeed fixed.

chrishoage, 2022-11-06:
@tzarc I can confirm that https://github.com/qmk/qmk_firmware/pull/18977 fixes this issue. I was able to rebase on top of develop - recompiled - and MO(1) is working as expected. 

Thank you for the quick fix! 

Closing as fixed.

PR #18977 (2022-11-06): Fix keycode parameter extraction to match the new DD keycodes
## Description

QMK has many keycodes with parameters, like `TO(layer)` or `MT(mod, kc)`. Add macros for extracting the values of those parameters from the keycode, so that the bit allocation for those parameters can be performed in a single place where those macros are defined; then change various code that was extracting the parameters manually to use those macros.

Macro names are based on the full `QK_xxx` range name; e.g., for `MO(layer)` it is `QK_MOMENTARY_GET_LAYER(kc)`. 

This change actually fixes several regressions intro [... cut]
commit a7b2f423 (2022-11-06): Fix keycode parameter extraction to match the new DD keycodes (#18977)

* Add macros to extract parameters from keycode values

Implement both encoding and decoding for keycodes like TO(layer) or
LM(layer, mod) in one place, so that the decoding won't get out of sync
with the encoding.

While at it, fix some macros for creating keycode values that did not
apply the appropriate masks to parameters (and therefore could allow the
result to be out of range if a wrong parameter was passed).

* keymap_common: Use extraction macros for keycodes

* pointing_device_auto_mouse: Use extraction macros for [... cut]
closed 2022-11-06 by chrishoage