qmk/qmk_firmware #18980: [Bug] Cirque mouse_report jumps when scale changes in absolute mode

chrishoage, 2022-11-07:
### Describe the Bug

When changing the scale/cpi while a touch is active may cause the `mouse_report` x/y values to jump by the maximum allowed constrained value. 

This is caused by the previous x/y values from touchData stored in `static uint16_t   x = 0, y = 0;` to have a different scale than the just applied `cirque_pinnacle_scale_data(&touchData, cirque_pinnacle_get_scale(), cirque_pinnacle_get_scale());` touchData value. When the operations outlined below are calculated it is calculated as a large movement (-128, -128) causing the cursor to "jump" as soon as the report is sent. 

```c
        if (x && y && touchData.xValue && touchData.yValue) {
            report_x = CONSTRAIN_HID_XY((int16_t)(touchData.xValue - x));
            report_y = CONSTRAIN_HID_XY((int16_t)(touchData.yValue - y));
        }
```

It seems as if the static x/y values also need to be scaled in order for this operation to actually compare the delta for the `mouse_report` however maybe there is another / better solution? 

I would be happy to try to tackle this myself - I just might need a hint on how this is best solved to set me in the right direction. 

### Keyboard Used

spleeb (not yet in qmk)

### Link to product page (if applicable)

https://github.com/chrishoage/spleeb

### Operating System

Linux

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /home/chris/projects/qmk_firmware
Ψ Detected Linux.
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32-dfu' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'md-boot' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Git branch: hoage/sctratchpad
Ψ Repo version: 0.18.16
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest hoage/sctratchpad: 2022-11-06 23:48:31 +0000 (442e7653cc) -- Fix use of shifted custom keycode (#18978)
Ψ - Latest upstream/master: 2022-11-05 10:55:42 -0700 (e3a97eb1c8) -- [Keyboard] Add Work Louder Numpad (#18912)
Ψ - Latest upstream/develop: 2022-11-06 23:48:31 +0000 (442e7653cc) -- Fix use of shifted custom keycode (#18978)
Ψ - Common ancestor with upstream/master: 2022-11-05 10:55:42 -0700 (e3a97eb1c8) -- [Keyboard] Add Work Louder Numpad (#18912)
Ψ - Common ancestor with upstream/develop: 2022-11-06 23:48:31 +0000 (442e7653cc) -- Fix use of shifted custom keycode (#18978)
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-09-18 10:01:17 +0000 --  (0e9d558b5)
Ψ - lib/chibios-contrib: 2022-10-03 18:09:41 +0200 --  (bb8356fb)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

I added extra logs in `pointing_device_drivers.c`  Here you can see the `x=-128 y=-128` report after the scale changes (`is_sniping_enabled`)

```
chrishoage:spleeb:1: cirque_pinnacle touchData x=1442 y= 682 z=26
chrishoage:spleeb:1: cirque_pinnacle touchData scaled x= 462 y= 276
chrishoage:spleeb:1: cirque_pinnacle report x=   0 y=   0
chrishoage:spleeb:1: cirque_pinnacle touchData x=1442 y= 682 z=28
chrishoage:spleeb:1: cirque_pinnacle touchData scaled x= 462 y= 276
chrishoage:spleeb:1: cirque_pinnacle report x=   0 y=   0
chrishoage:spleeb:1: cirque_pinnacle touchData x=1442 y= 682 z=27
chrishoage:spleeb:1: cirque_pinnacle touchData scaled x= 462 y= 276
chrishoage:spleeb:1: cirque_pinnacle report x=   0 y=   0
chrishoage:spleeb:1: (spleeb) process_record_kb: config = {
chrishoage:spleeb:1: 	raw = 0x80,
chrishoage:spleeb:1: 	{
chrishoage:spleeb:1: 		is_dragscroll_enabled=0
chrishoage:spleeb:1: 		is_sniping_enabled=1
chrishoage:spleeb:1: 		default_dpi=0x0 (400)
chrishoage:spleeb:1: 		sniping_dpi=0x0 (200)
chrishoage:spleeb:1: 	}
chrishoage:spleeb:1: }
chrishoage:spleeb:1: cirque_pinnacle touchData x=1442 y= 682 z=28
chrishoage:spleeb:1: cirque_pinnacle touchData scaled x= 231 y= 138
chrishoage:spleeb:1: cirque_pinnacle report x=-128 y=-128
chrishoage:spleeb:1: cirque_pinnacle touchData x=1442 y= 682 z=26
chrishoage:spleeb:1: cirque_pinnacle touchData scaled x= 231 y= 138
chrishoage:spleeb:1: cirque_pinnacle report x=   0 y=   0
```

Steps to reproduce this are:

1. Have Cirque in absolute mode
2. While touching the trackpad change the CPI to a different value 
3. Observe the cursor on screen jumps (-128, -128)

chrishoage, 2022-11-07:
I have very naively fixed it with this diff (please ignore the coef change it's unrelated to this - https://github.com/qmk/qmk_firmware/pull/18966)

I used the similar pattern as the static x,y delta compare values where I store the last scale value, and only send the report when the last scale and the current scale are the same. 

While this technically is causing a report to not get sent which would otherwise be sent it seemed better than the "jump". In practice while testing this I couldn't actually feel any mouse update being lost. I presume due the scan rate (I believe is 10ms). This approach may not be ideal if the cpi/scale changed very frequently, but I feel that isn't common.  

If this approach is okay I will open a PR

```diff
diff --git a/quantum/pointing_device/pointing_device_drivers.c b/quantum/pointing_device/pointing_device_drivers.c
index 172c9f36d7..20e5c3e957 100644
--- a/quantum/pointing_device/pointing_device_drivers.c
+++ b/quantum/pointing_device/pointing_device_drivers.c
@@ -107,8 +107,9 @@ void cirque_pinnacle_enable_cursor_glide(bool enable) {
     cursor_glide_enable = enable;
 }
 
-void cirque_pinnacle_configure_cursor_glide(float trigger_px) {
+void cirque_pinnacle_configure_cursor_glide(float trigger_px, float coef) {
     glide.config.trigger_px = trigger_px;
+    glide.config.coef       = coef;
 }
 #    endif
 
@@ -126,11 +127,13 @@ report_mouse_t cirque_pinnacle_get_report(report_mouse_t mouse_report) {
     pinnacle_data_t   touchData = cirque_pinnacle_read_data();
     mouse_xy_report_t report_x = 0, report_y = 0;
     static uint16_t   x = 0, y = 0;
+    static uint16_t   last_scale = 0;
+    uint16_t          scale      = cirque_pinnacle_get_scale();
 #        if defined(CIRQUE_PINNACLE_TAP_ENABLE)
-    mouse_report.buttons        = pointing_device_handle_buttons(mouse_report.buttons, false, POINTING_DEVICE_BUTTON1);
+    mouse_report.buttons         = pointing_device_handle_buttons(mouse_report.buttons, false, POINTING_DEVICE_BUTTON1);
 #        endif
 #        ifdef POINTING_DEVICE_GESTURES_CURSOR_GLIDE_ENABLE
-    cursor_glide_t glide_report = {0};
+    cursor_glide_t glide_report  = {0};
 
     if (cursor_glide_enable) {
         glide_report = cursor_glide_check(&glide);
@@ -157,15 +160,16 @@ report_mouse_t cirque_pinnacle_get_report(report_mouse_t mouse_report) {
 #        endif
 
     // Scale coordinates to arbitrary X, Y resolution
-    cirque_pinnacle_scale_data(&touchData, cirque_pinnacle_get_scale(), cirque_pinnacle_get_scale());
+    cirque_pinnacle_scale_data(&touchData, scale, scale);
 
     if (!cirque_pinnacle_gestures(&mouse_report, touchData)) {
-        if (x && y && touchData.xValue && touchData.yValue) {
+        if (last_scale && scale == last_scale && x && y && touchData.xValue && touchData.yValue) {
             report_x = CONSTRAIN_HID_XY((int16_t)(touchData.xValue - x));
             report_y = CONSTRAIN_HID_XY((int16_t)(touchData.yValue - y));
         }
-        x = touchData.xValue;
-        y = touchData.yValue;
+        x          = touchData.xValue;
+        y          = touchData.yValue;
+        last_scale = scale;
 
 #        ifdef POINTING_DEVICE_GESTURES_CURSOR_GLIDE_ENABLE
         if (cursor_glide_enable) {
```

drashna, 2022-12-10:
merged. Safe to say this can be closed.

closed 2022-12-10 by drashna