qmk/qmk_firmware #19008: [Bug] Some rp2040 boards will not wake host from sleep on keypress

chrishoage, 2022-11-09:
### Describe the Bug

I am working with the [Boardsource Blok](https://boardsource.xyz/store/628b95b494dfa308a6581622) for a project and I am having an issue where the keyboard will not wake the host from suspend-to-ram. 

This is similar in nature to https://github.com/qmk/qmk_firmware/issues/16934 though my board will continue to work after the computer wakes from suspend. 

To eliminate any variables with my code I used `handwired/onekey/rp2040` with the `default` keymap on a spare Blok I have and tried to replicate the issue. Even with this example code I still am unable to wake my computer from sleep. 

I have tried both my Lenovo Thinkpad X1 Carbon and my desktop (Gigabyte Z390). I have verified that usb wakeup is enabled for the usb devices. 

I have ordered a USB to TTL Module Serial Converter to be able to debug further, as `qmk console` isn't very useful while the computer is suspended :sweat_smile: 

I have a suspicion that the MCU never enters the `USB_DEVICE_STATE_SUSPEND` state and thus the `usbWakeupHost`  block never runs. However at this point I'm just guessing with out being able to debug the MCU state further while the host is suspended. 

```c
            if (suspend_wakeup_condition()) {
                usbWakeupHost(&USB_DRIVER);
                restart_usb_driver(&USB_DRIVER);
            }
```

If there are any hints about where to start looking for debugging this it would be very helpful! 

### Keyboard Used

handwired/onekey/rp2040

### Link to product page (if applicable)

https://boardsource.xyz/store/628b95b494dfa308a6581622

### Operating System

Linux

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /home/chris/projects/qmk_firmware
Ψ Detected Linux.
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32-dfu' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'md-boot' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/chris/projects/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Git branch: hoage/sctratchpad
Ψ Repo version: 0.18.16
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest hoage/sctratchpad: 2022-11-06 23:48:31 +0000 (442e7653cc) -- Fix use of shifted custom keycode (#18978)
Ψ - Latest upstream/master: 2022-11-05 10:55:42 -0700 (e3a97eb1c8) -- [Keyboard] Add Work Louder Numpad (#18912)
Ψ - Latest upstream/develop: 2022-11-07 15:16:05 -0700 (ce80a3ad46) -- Fixup hotdox76v2 on develop (#18991)
Ψ - Common ancestor with upstream/master: 2022-11-05 10:55:42 -0700 (e3a97eb1c8) -- [Keyboard] Add Work Louder Numpad (#18912)
Ψ - Common ancestor with upstream/develop: 2022-11-06 23:48:31 +0000 (442e7653cc) -- Fix use of shifted custom keycode (#18978)
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-09-18 10:01:17 +0000 --  (0e9d558b5)
Ψ - lib/chibios-contrib: 2022-10-03 18:09:41 +0200 --  (bb8356fb)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

chrishoage, 2022-12-02:
I am no longer experiencing this issue. I think perhaps a kernel update fixed this for me. 

Closing for now, will re-open if it reappears.

closed 2022-12-02 by chrishoage

lorenSchmidt, 2024-07-21:
i appear to be having the same or a similar problem. on sleeping the computer, it will not wake on keypress. additionally, i cannot type my password to log in once i have woken it. the keyboard works fine if i unplug it and plug it back in, suggesting it's a state issue fixed by power cycling. sofle rgb, latest version bundled with QMK MSYS.