qmk/qmk_firmware #19046: [Bug] MIDI data not received by keyboard

samesimilar, 2022-11-13:
### Describe the Bug

I have enabled MIDI in my firmware build, and I can successfully send MIDI from my keyboard over USB to my computer.
I also expect to be able to send MIDI from my computer to the keyboard.
- The keyboard shows up as a MIDI output device on my computer
- I have registered MIDI callbacks for receiving data.
- MIDI_ENABLE = yes in my rules.mk

Observed:
The callback function `usb_get_midi` never receives MIDI data from the `recv_midi_packet` function. 

### Keyboard Used

wilba_tech/rama_works_m50_a

### Operating System

macOS 12.5.1

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

I was able to resolve the bug  for myself by changing the following line in tmk_core/protocol/lufa/lufa.c from:

`ConfigSuccess &= Endpoint_ConfigureEndpoint((MIDI_STREAM_OUT_EPNUM | ENDPOINT_DIR_IN), EP_TYPE_BULK, MIDI_STREAM_EPSIZE, 1);`

to: 

`ConfigSuccess &= Endpoint_ConfigureEndpoint((MIDI_STREAM_OUT_EPNUM | ENDPOINT_DIR_OUT), EP_TYPE_BULK, MIDI_STREAM_EPSIZE, 1);`

It seems to me that the incorrect ENDPOINT direction was selected in this configuration in the upstream codebase. Changing it makes MIDI input work and I can now process MIDI on the keyboard. But I am not sure if the original configuration was intentional or a bug, so I wanted to share it in case someone else has the same problem as me!

zvecr, 2022-11-13:
Already fixed by #18654

closed 2022-11-13 by zvecr

qiji102, 2024-01-16:
I am currently able to send MIDI information from the keyboard to the computer, but now I want to send data from the computer to the keyboard to control RGB lights. The documentation for MIDI reception in QMK is insufficient, and I would like to know how to implement the code for receiving and processing MIDI data.

nezumee, 2024-02-16:
> I am currently able to send MIDI information from the keyboard to the computer, but now I want to send data from the computer to the keyboard to control RGB lights.

You can try something like https://github.com/nezumee/qmk_firmware/tree/4x4/keyboards/4x4/keymaps/midi-controller (code is a bit old, hopefully it still works). The code for processing midi events is in midi-feedback.c, setting RGB LEDs is in rgb_matrix_user.inc.