qmk/qmk_firmware #19071: [Bug] Can't build qmk, printf macross issue

stalkerg, 2022-11-14:
### Describe the Bug

Just trying to build it, I see this errors:
```
Compiling: platforms/chibios/drivers/eeprom/eeprom_stm32.c                                         In file included from /usr/arm-none-eabi/include/stdio.h:800,
                 from platforms/chibios/drivers/eeprom/eeprom_stm32.c:21:
lib/printf/src/printf/printf.h:71:21: error: expected declaration specifiers or '...' before numeric constant
   71 | # define sprintf_   sprintf
      |                     ^~~~~~~
lib/printf/src/printf/printf.h:144:6: note: in expansion of macro 'sprintf_'
  144 | int  sprintf_(char* s, const char* format, ...) ATTR_PRINTF(2, 3);
      |      ^~~~~~~~
In file included from /usr/arm-none-eabi/include/ssp/stdio.h:34,
                 from /usr/arm-none-eabi/include/stdio.h:800,
                 from platforms/chibios/drivers/eeprom/eeprom_stm32.c:21:
lib/printf/src/printf/printf.h:71:21: error: expected declaration specifiers or '...' before '__builtin_object_size'
   71 | # define sprintf_   sprintf
      |                     ^~~~~~~
lib/printf/src/printf/printf.h:144:6: note: in expansion of macro 'sprintf_'
  144 | int  sprintf_(char* s, const char* format, ...) ATTR_PRINTF(2, 3);
      |      ^~~~~~~~
In file included from /usr/arm-none-eabi/include/stdio.h:800,
                 from platforms/chibios/drivers/eeprom/eeprom_stm32.c:21:
lib/printf/src/printf/printf.h:72:21: error: expected declaration specifiers or '...' before numeric constant
   72 | # define vsprintf_  vsprintf
      |                     ^~~~~~~~
lib/printf/src/printf/printf.h:146:5: note: in expansion of macro 'vsprintf_'
  146 | int vsprintf_(char* s, const char* format, va_list arg) ATTR_VPRINTF(2);
      |     ^~~~~~~~~
In file included from /usr/arm-none-eabi/include/ssp/stdio.h:34,
                 from /usr/arm-none-eabi/include/stdio.h:800,
                 from platforms/chibios/drivers/eeprom/eeprom_stm32.c:21:
lib/printf/src/printf/printf.h:72:21: error: expected declaration specifiers or '...' before '__builtin_object_size'
   72 | # define vsprintf_  vsprintf
      |                     ^~~~~~~~
lib/printf/src/printf/printf.h:146:5: note: in expansion of macro 'vsprintf_'
  146 | int vsprintf_(char* s, const char* format, va_list arg) ATTR_VPRINTF(2);
      |     ^~~~~~~~~
In file included from /usr/arm-none-eabi/include/stdio.h:800,
                 from platforms/chibios/drivers/eeprom/eeprom_stm32.c:21:
lib/printf/src/printf/printf.h:73:21: error: expected declaration specifiers or '...' before numeric constant
   73 | # define snprintf_  snprintf
      |                     ^~~~~~~~
lib/printf/src/printf/printf.h:167:6: note: in expansion of macro 'snprintf_'
  167 | int  snprintf_(char* s, size_t count, const char* format, ...) ATTR_PRINTF(3, 4);
      |      ^~~~~~~~~
In file included from /usr/arm-none-eabi/include/ssp/stdio.h:34,
                 from /usr/arm-none-eabi/include/stdio.h:800,
                 from platforms/chibios/drivers/eeprom/eeprom_stm32.c:21:
lib/printf/src/printf/printf.h:73:21: error: expected declaration specifiers or '...' before '__builtin_object_size'
   73 | # define snprintf_  snprintf
      |                     ^~~~~~~~
lib/printf/src/printf/printf.h:167:6: note: in expansion of macro 'snprintf_'
  167 | int  snprintf_(char* s, size_t count, const char* format, ...) ATTR_PRINTF(3, 4);
      |      ^~~~~~~~~
In file included from /usr/arm-none-eabi/include/stdio.h:800,
                 from platforms/chibios/drivers/eeprom/eeprom_stm32.c:21:
lib/printf/src/printf/printf.h:74:21: error: expected declaration specifiers or '...' before numeric constant
   74 | # define vsnprintf_ vsnprintf
      |                     ^~~~~~~~~
lib/printf/src/printf/printf.h:169:5: note: in expansion of macro 'vsnprintf_'
  169 | int vsnprintf_(char* s, size_t count, const char* format, va_list arg) ATTR_VPRINTF(3);
      |     ^~~~~~~~~~
In file included from /usr/arm-none-eabi/include/ssp/stdio.h:34,
                 from /usr/arm-none-eabi/include/stdio.h:800,
                 from platforms/chibios/drivers/eeprom/eeprom_stm32.c:21:
lib/printf/src/printf/printf.h:74:21: error: expected declaration specifiers or '...' before '__builtin_object_size'
   74 | # define vsnprintf_ vsnprintf
      |                     ^~~~~~~~~
lib/printf/src/printf/printf.h:169:5: note: in expansion of macro 'vsnprintf_'
  169 | int vsnprintf_(char* s, size_t count, const char* format, va_list arg) ATTR_VPRINTF(3);
      |     ^~~~~~~~~~
 [ERRORS]
 | 
 | 
 | 
gmake[1]: *** [builddefs/common_rules.mk:359: .build/obj_preonic_rev3_drop_default/eeprom_stm32.o] Error 1
gmake: *** [Makefile:414: preonic/rev3_drop:default] Error 1
```

Gcc for the arm was installed by `crossdev --stable -t arm-none-eabi -s4`

### Keyboard Used

preonic/rev3_drop

### Operating System

Gentoo Linux

### qmk doctor Output

```text
QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /home/stalkerg/qmk_firmware
Ψ Detected Linux.
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp /home/stalkerg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp /home/stalkerg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32-dfu' boards. Run 'sudo cp /home/stalkerg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp /home/stalkerg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp /home/stalkerg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp /home/stalkerg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /home/stalkerg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp /home/stalkerg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'md-boot' boards. Run 'sudo cp /home/stalkerg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp /home/stalkerg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/stalkerg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Git branch: master
Ψ Repo version: 0.18.17
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2022-11-13 15:48:31 -0700 (ee7426e7b1) -- Fixup day_off (#19067)
Ψ - Latest upstream/master: 2022-11-13 15:48:31 -0700 (ee7426e7b1) -- Fixup day_off (#19067)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2022-11-13 15:48:31 -0700 (ee7426e7b1) -- Fixup day_off (#19067)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 11.3.0
Ψ Found avr-gcc version 13
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-06-11 09:09:45 +0000 --  (f836d24b0)
Ψ - lib/chibios-contrib: 2022-07-27 10:46:15 +0200 --  (d03aa9cc)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-05-17 19:19:01 +0200 --  (07edde8)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

waffle87, 2022-11-14:
`-lc` & `-specs=nosys.specs` cflags seems to "fix" this — so `qmk compile -kb preonic/rev3_drop -km default -e CFLAGS+="-lc -specs=nosys.specs"`
I suggest joining the QMK discord server and reading through [this thread](https://discord.com/channels/440868230475677696/1023411182926827531/1024773136924815440).

stalkerg, 2022-11-14:
@waffle87 seems like QMK doesn't support the latest GCC, am I right? Do you have a ticket for such an issue?

stalkerg, 2022-11-15:
@waffle87 no, it's gcc 11

stalkerg, 2022-11-17:
Fixed by:
```
qmk compile -kb preonic/rev3_drop -km default -e EXTRAFLAGS+="-U_FORTIFY_SOURCE"
```

It's needed because Gentoo GCC enables, by default, the SSP (FORTIFY level 2) for any tasks with optimization. 
Current QMK printf implementation doesn't support SSP version.