qmk/qmk_firmware #19092: cant compile and error doesn't make sense

thecodsman, 2022-11-16:
### Issue Description

[image]
I dont understand this error what expression does it want ;-;

also heres the file
[image]
please help

fauxpark, 2022-11-16:
You haven't shown the entire output.

thecodsman, 2022-11-16:
[image]

here you go

fauxpark, 2022-11-16:
You should probably also post your keymap, and make sure there is no whitespace after the backslashes in the layout macro.

thecodsman, 2022-11-16:
here is the keymap
[image]
but wdym by the layout macro? the one in ansi.h?

thecodsman, 2022-11-16:
wait

fauxpark, 2022-11-17:
The layer names need to be wrapped in `[]`:
[image]

thecodsman, 2022-11-17:
lets goooooo it worked. thank you so much

closed 2022-11-17 by zvecr