qmk/qmk_firmware #19095: [Bug]  Cannot flash on Alpine linux

timsofteng, 2022-11-16:
### Describe the Bug

I cannot flash the but on alpine linux. I've installed all necessary deps and copied udev rules.
Then reboot to make sure udev rules bind.

But after reseting the board I've got that loading
```
 * The firmware size is fine - 26998/28672 (94%, 1674 bytes free)
Flashing for bootloader: caterina
Waiting for USB serial port - reset your controller now (Ctrl+C to cancel)........
Device /dev/ttyACM0 has appeared; assuming it is the controller.
Waiting for /dev/ttyACM0 to become writable...........................
```
which spins forever.

### Keyboard Used

qmk flash -kb crkbd -km default

### Operating System

Alpine Linux (edge)

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /home/tim/qmk_firmware
Ψ Detected Linux.
Ψ Git branch: master
Ψ Repo version: 0.18.17
Ψ - Latest master: 2022-11-16 02:40:28 +0800 (df5689ed06) -- [Docs] Note default value for TAP_CODE_DELAY (#19078)
Ψ - Latest upstream/master: 2022-11-16 16:07:51 +0900 (d6956668e3) -- Specify `DEFAULT_FOLDER` for `v2/splinky` (#19093)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2022-11-16 02:40:28 +0800 (df5689ed06) -- [Docs] Note default value for TAP_CODE_DELAY (#19078)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-06-11 09:09:45 +0000 --  (f836d24b0)
Ψ - lib/chibios-contrib: 2022-07-27 10:46:15 +0200 --  (d03aa9cc)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-05-17 19:19:01 +0200 --  (07edde8)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
> qmk doctor > wl-copy
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /home/tim/qmk_firmware
Ψ Detected Linux.
Ψ Git branch: master
Ψ Repo version: 0.18.17
Ψ - Latest master: 2022-11-16 02:40:28 +0800 (df5689ed06) -- [Docs] Note default value for TAP_CODE_DELAY (#19078)
Ψ - Latest upstream/master: 2022-11-16 16:07:51 +0900 (d6956668e3) -- Specify `DEFAULT_FOLDER` for `v2/splinky` (#19093)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2022-11-16 02:40:28 +0800 (df5689ed06) -- [Docs] Note default value for TAP_CODE_DELAY (#19078)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-06-11 09:09:45 +0000 --  (f836d24b0)
Ψ - lib/chibios-contrib: 2022-07-27 10:46:15 +0200 --  (d03aa9cc)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-05-17 19:19:01 +0200 --  (07edde8)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

drashna, 2022-12-10:
is modem manager disabled/reconfigured probably?

orgo89, 2023-04-01:
Same for Fedora 37

sigprof, 2023-04-01:
> I cannot flash the keyboard on alpine linux. I've installed all necessary deps and copied udev rules.

Are you actually using `udev` on your Alpine Linux installation?  Apparently it might not be the default, and you may need to run `setup-devd udev` to switch to `udev` instead of `mdev`.

And yet another issue is that the udev rules supplied by QMK use `TAG+="uaccess"` to give permissions to the currently active user, but that feature requires either the full-sized `systemd` or the alternative `elogind` implementation of session management, and the `elogind` package might not be installed either (users of Void Linux often encounter the same issue).  According to https://wiki.alpinelinux.org/wiki/Elogind, the way to install `elogind` on Alpine Linux properly is:
```
# apk add elogind polkit-elogind
# rc-update add elogind
# rc-service elogind start
```

sigprof, 2023-04-01:
> Same for Fedora 37

@orgo89, your issue with Fedora 37 is probably not the same as this — please open a separate issue with complete information as requested in the template.

orgo89, 2023-04-01:
Not totally sure whether it is related or not but was solved by installing and executing all the commands as super user.

tzarc, 2023-04-01:
> Not totally sure whether it is related or not but was solved by installing and executing all the commands as super user.

For anyone else coming across this issue, please don't do this. `sudo` is not the answer -- the installation scripts already elevate permissions when actually necessary.

If you use `sudo` to do such things, you end up with:
* files in your home directory owned by `root` and thus unable to be deleted by your normal user
* future upgrade fail due to a lack of write permissions

In @orgo89's case it's likely `sudo` was previously used, and now upgrades fail. I personally use Fedora 37 and it works perfectly fine.

levon-dalakyan, 2024-08-17:
i have same problem on mac