qmk/qmk_firmware #19157: [BUG] IO error Can't flash keyboard with qmk toolbox on macos Ventura

plantyplantman, 2022-11-26:
### Describe the Bug

I am unable to flash KBD67 mkiirgb v3  with QMK toolbox since upgrading to MacOS ventura.

When I try to flash my keyboard, it comes up with the IO error
~IO ERROR: “kbdfans_kbd67_mkiirgb_v3_333.bin” couldn’t be copied to “KBDFANS  ”.~

I am able to flash the same .bin file on MacOS moneterey with no issues.

[image]

### Keyboard Used

Kbdfans/kbd67/mkiirgb/v3

### Operating System

MacOS ventura 13.0.1

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /Users/home/qmk_firmware
Ψ Detected macOS 13.0.1 (Apple Silicon).
Ψ Git branch: master
Ψ Repo version: 0.18.17
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2022-11-21 12:17:09 +0800 (9291c8f867) -- [Keyboard] add og60 keyboard (#19114)
Ψ - Latest upstream/master: 2022-11-26 04:31:58 +1100 (8f9b49dc5b) -- Fix build failures for `bastardkb/tbk` and `jels/boaty` (#19152)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2022-11-21 12:17:09 +0800 (9291c8f867) -- [Keyboard] add og60 keyboard (#19114)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-06-11 09:09:45 +0000 --  (f836d24b0)
Ψ - lib/chibios-contrib: 2022-07-27 10:46:15 +0200 --  (d03aa9cc)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-05-17 19:19:01 +0200 --  (07edde8)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

Via

filterpaper, 2022-11-27:
This might be related to [Finder issues with UF2](https://blog.adafruit.com/2022/10/31/uploading-uf2-files-with-macos-13-0-ventura-apple-microbit_edu-raspberry_pi-circuitpython/). Does it work with one of the following?

1. Copy using the command prompt: `cp kbdfans_kbd67_mkiirgb_v3_333.bin /Volumes/KBDFANS/FLASH.BIN`.
2. Use a third party file manager (such as Command One or Marta) to replace `FLASH.BIN` with the new firmware file.

plantyplantman, 2022-12-01:
Dang.. I really hoped it wasn't an apple issue.

I tried copying via terminal and using Marta but neither worked.

:(

Guess we just have to cross our fingers and hope that apple will do something

brunzefb, 2022-12-15:
@plantyplantman I am using Ventura, running the QMK toolbox and programming my keyboard without problems. It may not be an OS related problem, rather a local corruption on your machine. Are you sure your /Volumes/KDBFANS is mounted read/write, and has space (df -h /Volumes/KBDFANS ) is your friend.  Can you manually copy a file to the target?

brymon68, 2023-01-03:
Having a similar issue. QMK Toolbox gives me the following error: 
```
IO ERROR: “dztech_dz60rgb_wkl_v2_1_wkl_dz_qmk_v4.bin” couldn’t be copied to “KBDFANS  ”.
```
Tried manually copying: 
```
cp dztech_dz60rgb_wkl_v2_1_wkl_dz_qmk_v6.bin /Volumes/KBDFANS\ \ /FLASH.BIN
```

df -h
```
└─ $ df -h
Filesystem                Size   Used  Avail Capacity iused      ifree %iused  Mounted on
/dev/disk3s1s1           926Gi  8.2Gi  707Gi     2%  348618 4291206242    0%   /
devfs                    347Ki  347Ki    0Bi   100%    1200          0  100%   /dev
/dev/disk3s6             926Gi  2.0Gi  707Gi     1%       2 7412593560    0%   /System/Volumes/VM
/dev/disk3s2             926Gi  4.4Gi  707Gi     1%     858 7412593560    0%   /System/Volumes/Preboot
/dev/disk3s4             926Gi   11Mi  707Gi     1%      58 7412593560    0%   /System/Volumes/Update
/dev/disk1s2             500Mi  6.0Mi  482Mi     2%       3    4936040    0%   /System/Volumes/xarts
/dev/disk1s1             500Mi  6.1Mi  482Mi     2%      31    4936040    0%   /System/Volumes/iSCPreboot
/dev/disk1s3             500Mi  1.1Mi  482Mi     1%      34    4936040    0%   /System/Volumes/Hardware
/dev/disk3s5             926Gi  204Gi  707Gi    23% 1315474 7412593560    0%   /System/Volumes/Data
map auto_home              0Bi    0Bi    0Bi   100%       0          0  100%   /System/Volumes/Data/home
msdos://disk4/KBDFANS     42Ki   38Ki  4.0Ki    91%       1          0  100%   /Volumes/KBDFANS
```

alzedd, 2023-07-08:
any news about this issue?
I'm facing the exact same problem and since I don't have access to a windows or linux pc I can't update my qmk configs :/

sigprof, 2023-07-08:
You may try using the `dd` command to overwrite the existing `FLASH.BIN` file in place — a similar solution works in Linux, but it requires several options which the MacOS implementation of `dd` apparently does not have, so there is some chance that this won't actually work:
```sh
dd conv=notrunc if=my_new_firmware.bin of=/Volumes/KBDFANS\ \ /FLASH.BIN
```

alzedd, 2023-07-10:
well, with brew I can get `gdd` which is the GNU implementation of `dd`... it could work :) 
Thank you! @sigprof, I'm going to try later