qmk/qmk_firmware #19184: [Bug]

nucatus, 2022-11-28:
### Describe the Bug

My aim is to invert the shift layer for the numerical keys on an ANSI layout. For instance, when the `1` key is pressed, `!` is output, and when `shift + 1` is pressed, `1` is output.

```
const key_override_t ko_1 = ko_make_with_layers(0, KC_1, KC_EXLM, 1 << _DVP_LAYER);       // 1 is !
const key_override_t ko_sh_1 = {                                                          // shift + 1 is 1
                                   .trigger_mods           = MOD_MASK_SHIFT,
                                   .layers                 = 1 << _DVP_LAYER,
                                   .suppressed_mods        = MOD_MASK_SHIFT,
                                   .options                = ko_options_default,
                                   .trigger                = KC_1,
                                   .replacement            = KC_1,
                                   .enabled                = NULL};
const key_override_t **key_overrides = (const key_override_t *[]){
  &ko_1,
  &ko_sh_1,
  NULL
};

```

The first override works as expected (when `1` is pressed, `!` is output), but wher `shift + 1` is pressed `!` is output as well, but `1` was expected.

The karabiner event viewer shows the following output for the `ko_1` override:
```
[
  {
    "type": "down",
    "name": {"key_code":"left_shift"},
    "usagePage": "7 (0x0007)",
    "usage": "225 (0x00e1)",
    "misc": "flags left_shift"
  },
  {
    "type": "down",
    "name": {"key_code":"1"},
    "usagePage": "7 (0x0007)",
    "usage": "30 (0x001e)",
    "misc": "flags left_shift"
  },
  {
    "type": "up",
    "name": {"key_code":"1"},
    "usagePage": "7 (0x0007)",
    "usage": "30 (0x001e)",
    "misc": "flags left_shift"
  },
  {
    "type": "up",
    "name": {"key_code":"left_shift"},
    "usagePage": "7 (0x0007)",
    "usage": "225 (0x00e1)",
    "misc": ""
  }
]
```

and the following output in the case of the `ko_sh_1` override:
```
[
  {
    "type": "down",
    "name": {"key_code":"left_shift"},
    "usagePage": "7 (0x0007)",
    "usage": "225 (0x00e1)",
    "misc": "flags left_shift"
  },
  {
    "type": "down",
    "name": {"key_code":"1"},
    "usagePage": "7 (0x0007)",
    "usage": "30 (0x001e)",
    "misc": "flags left_shift"
  },
  {
    "type": "up",
    "name": {"key_code":"1"},
    "usagePage": "7 (0x0007)",
    "usage": "30 (0x001e)",
    "misc": "flags left_shift"
  },
  {
    "type": "up",
    "name": {"key_code":"left_shift"},
    "usagePage": "7 (0x0007)",
    "usage": "225 (0x00e1)",
    "misc": ""
  }
]
```

### Keyboard Used

durgod/k3x0/k320

### Operating System

OSX

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [X] Karabiner (macOS)

nucatus, 2022-11-28:
I found a reddit thread where the issue was discussed and the fix was to add a custom action to flush the modifiers.

https://www.reddit.com/r/olkb/comments/w389av/comment/igyvyrx/

Is this the intended behaviour?

drashna, 2022-12-10:
Tagging the original author: @JonasGessner

JonasGessner, 2022-12-10:
@nucatus your overrides look fine on first glance. If I remember correctly, people have reported issues before with Key Overrides and just having Karabiner installed, specifically with modifiers not getting updated properly. Can you try quitting Karabiner to see if that helps?

nucatus, 2023-01-13:
@JonasGessner, indeed, closing karabiner solved the issue. Thanks for the tip.

closed 2023-01-13 by nucatus