qmk/qmk_firmware #19254: [Bug] Key behave as if it was stuck and repeats the last typed letter

jakubgros, 2022-12-07:
### Describe the Bug

When typing, the keyboard sometimes behaves as if a key was pressed and held. It sends a signal to the PC until I click any key or disconnect a receiver. I think it's not a hardware problem because I've tried turning off the keyboard once the bug appeared and it was still sending the signal to PC until I disconnected a receiver.

The bug is hard to reproduce and I'm not sure what causes it. I think that it might be related to pressing multiple keys at the same time. What's interesting, is that my keyboard sometimes works perfectly and then it gets into a state where it's more likely to enter the bug. Restarting the keyboard/receiver doesn't help to get out of the "increased likelihood of bug state". After some time it gets fixed by itself.

Right before the bug is triggered, when I'm typing, it hangs for ~1s(nothing is typed into PC even though I'm pressing keys) and it starts repeating the last pressed key.

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

getreuer, 2022-12-07:
> Right before the bug is triggered, when I'm typing, it hangs for ~1s(nothing is typed into PC even though I'm pressing keys) and it starts repeating the last pressed key.

Just to clarify, the keyboard is connected to the computer wirelessly over Bluetooth, is that right? I wonder if the stuck keys are the result of Bluetooth communication occasionally getting interrupted, just after a key press event and the computer missing the corresponding key release event.

jakubgros, 2022-12-08:
Yes, it's connected wirelessly to the computer. [This is the module I have that is used as a receiver](https://falba.tech/product/yj-14015-module-core51822b-clone/). Based on what I found on the internet it's a Bluetooth module. Though, I'm not 100% sure how it works, because I'm not that much into hardware and I'm confused why an additional Bluetooth module had to be added, instead of using the one from laptop.

The connection scheme looks as follows:
```
-- = cable
.... = wireless connection
PC ------receiver . . . . . . . . . . . . . . . . . . . . . .  \/ . . . . . . . . . . \/    
                                                           left keyboard      right keyboard
````

fauxpark, 2022-12-08:
This board does not use Bluetooth, all three modules communicate over the Gazell protocol, and the receiver module transmits the key state to the actual MCU via UART.

https://github.com/mattdibi/redox-w-firmware
https://github.com/qmk/qmk_firmware/blob/master/keyboards/redox_w/matrix.c

jakubgros, 2022-12-15:
I managed to gather a bit more useful information about the problem. When my keyboard got into "increased likelihood of bug state", I tried to restart the keyboard but it didn't help. Then I plugged it into another computer and it worked there without any problem. I could still observe the problematic behaviour when I plugged it back into the initial PC. Based on that, I assume that the problem might be on the PC side of communication.

korompaiistvan, 2022-12-27:
I also ran into this issue on a kbd67rev2 board but could not figure out what was causing it. the frequency of this bug seemed to correlate with my machine's CPU load

sneakret, 2023-02-01:
I am seeing the exact same symptoms (keyboard events lost when CPU load is high -- input pausing, then acting like a key is being held down, then dropping a couple keystrokes, then resuming normal function). Like @jakubgros I can't reproduce it on my other computer. I have ruled out QMK as the cause in my environment, because I connected a non-QMK keyboard and the problem persisted.

mrschyte, 2023-12-21:
I'm having a similar issue when my keyboard is connected through a KVM switch. I have set TAP_CODE_DELAY to 10ms which greatly reduced the frequency of this happening. I wonder if the same workaround would apply here as well.

mrschyte, 2023-12-28:
> I'm having a similar issue when my keyboard is connected through a KVM switch. I have set TAP_CODE_DELAY to 10ms which greatly reduced the frequency of this happening. I wonder if the same workaround would apply here as well.

I was able to completely resolve my issue by setting a higher `USB_POLLING_INTERVAL_MS`. My KVM seems to support only a 125Hz polling rate (8ms). The default 1000Hz polling rate results in dropped packets and if the dropped packet corresponds to a key-up event, it acts as if the key was stuck.

fauxpark, 2023-12-28:
That didn't do anything. Due to how USB works fundamentally, the value of `USB_POLLING_INTERVAL_MS` is more of a suggestion to the host; the keyboard does not actually send data faster or slower unless the host requests it more or less frequently.

mrschyte, 2023-12-31:
It did work in my case. I guess the KVM allows a higher polling interval to be negotiated than it can support.

niscient-green, 2024-08-06:
Had the same issue. SplitKB Kyria rev3 on QMK attached to KVM. Putting the line below in config.h worked.
#define USB_POLLING_INTERVAL_MS 10

Adrian-Grimm, 2024-08-13:
Same bug here with a KBD fans KBD8X MKII behind a KVM with VIA Firmware - I upgraded from an old version from 2021 and now I have this bug. Same with most recent QMK.

MrWolf-K9, 2024-09-27:
I thought I had the same issue but I simply forgot to call unregister_code() after register_code() in my custom macro.

j3rwin, 2025-08-19:
Same issue with Monsgeek M1 V3 wired version connected to my monitor's (Dell UP2414Q) USB hub. No idea on how to debug as it's very random.