qmk/qmk_firmware #19272: [Feature Request] SOLID_SPLASH underglow affect enhancement

LeoLinny, 2022-12-09:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Was playing around with the underglow effects on my Keychron Q3.

The effect currently refreshes with each key press, might be a personal taste, but coming from my old Keychron K2, the splash effect continued per key press, so to me, it looked better.

drashna, 2022-12-10:
Have you tried using the "multi splash"?  that should do what you're looking for.

LeoLinny, 2022-12-11:
That mode doesn't show up on my Keychron Q3

drashna, 2022-12-15:
Yeah, looks like it's explicitly not enabled: 
https://github.com/qmk/qmk_firmware/blob/33cc81f36efd255447b47428f3198ef3793c1d41/keyboards/keychron/q3/config.h#L104

If you are compiling from source, you could add that to your keymap's config.h to enabled.

closed 2023-04-16 by github-actions[bot]