qmk/qmk_firmware #19287: [Bug] Feker IK75 wrong status LED configuration

jestercore, 2022-12-10:
### Describe the Bug

Status LEDs for Windows key lock and Scroll lock are messed up and should be swapped.
At the moment when you lock Windows key - LED with "S" lights up. The same when you Scroll lock lights up "W".
That's a clear mistake.

Could you please fix it?  

### Keyboard Used

feker/ik75

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

https://github.com/qmk/qmk_firmware/tree/master/keyboards/feker/ik75

HorrorTroll, 2022-12-10:
can u specific more issue? It was running fine on my friend keebs and other people.

jestercore, 2022-12-10:
Wrong LEDs activating for Win Lock and Scroll Lock, they are swapped out.

https://photos.app.goo.gl/WtCdnj2y5uJvJ93b9

jestercore, 2022-12-10:
> can u specific more issue? It was running fine on my friend keebs and other people.

Wrong LEDs activating for Win Lock and Scroll Lock, they are swapped out.

https://photos.app.goo.gl/WtCdnj2y5uJvJ93b9

HorrorTroll, 2022-12-10:
[image]
oh about that one, it is Feker issue not the firmware wrong anyway. They just messed up label on the top case

jestercore, 2022-12-10:
> [image] oh about that one, it is Feker issue not the firmware wrong anyway. They just messed up label on the top case

Yes, but I think labels on case are prioritized over PCB markings.
How can it be swapped in firmware? Can I edit it and build new .hex firmware? Could you please explain how?

HorrorTroll, 2022-12-10:
uhhh, pcb marking was been used for default. The issue on top case is problem of them, there no way to fix this.

jestercore, 2022-12-10:
> uhhh, pcb marking was been used at default for indicator. The issue on top case is problem of them, there no way to fix this.

Is this really not possible to swap these LEDs in firmware? Like change two values?

HorrorTroll, 2022-12-10:
nope, it not how that work anyway, sorry.

jestercore, 2023-04-09:
I was managed to fix it by swapping colors for indicators:

before
```
bool rgb_matrix_indicators_user(void) {
    rgb_matrix_set_color(46, 0, 0, 0);
    rgb_matrix_set_color(104, 0, 0, 0);

    uint8_t red = host_keyboard_led_state().caps_lock ? 255 : 0;
    uint8_t _green_ = host_keyboard_led_state().scroll_lock ? 255 : 0;
    uint8_t _blue_ = keymap_config.no_gui ? 255 : 0;
```

after
```
bool rgb_matrix_indicators_user(void) {
    rgb_matrix_set_color(46, 0, 0, 0);
    rgb_matrix_set_color(104, 0, 0, 0);

    uint8_t red = host_keyboard_led_state().caps_lock ? 255 : 0;
    uint8_t **blue** = host_keyboard_led_state().scroll_lock ? 255 : 0;
    uint8_t **green** = keymap_config.no_gui ? 255 : 0;
```

Could you please merge this in the master?

jestercore, 2023-04-09:
> before
>     uint8_t green = host_keyboard_led_state().scroll_lock ? 255 : 0;
>     uint8_t blue = keymap_config.no_gui ? 255 : 0;
> 
> after
> 
>     uint8_t red = host_keyboard_led_state().caps_lock ? 255 : 0;
>     uint8_t **blue** = host_keyboard_led_state().scroll_lock ? 255 : 0;
>     uint8_t **green** = keymap_config.no_gui ? 255 : 0;
> 
> Could you please merge this in the master?

@HorrorTroll

Siioh, 2023-05-04:
Thank you for providing this solution.  I had the same problem, and it looks like it was still not yet applied to the master.  I was able to resolve it using your fix.