qmk/qmk_firmware #19295: [Bug] Keychron Q4 repeating keys leading to unresponsive board

sajenim, 2022-12-11:
### Describe the Bug

fa56e51894cf42cf0bc5bc4bce1b6f0d85c9dc34 Upon flashing the q4 the last pressed key repeats itself indefinitely and the board becomes unresponsive to further input requiring pulling the power cable. The issue presents itself within a dozen key presses. 

6f1c54d9dcd994d71bda95962bf9aa7049b8e404 is the last known working commit I have tested.

7ecb576a75fbe31c29a12a2fd478449349b89f7a may have introduced the issue.

Enabling LTO fixes this issue, I am unsure if this is the correct solution or if there is something deeper to resolve.

### Keyboard Used

keychron/q4/ansi_v2

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /home/minnie/.build/qmk_firmware
Ψ Detected Linux.
Ψ Git branch: master
Ψ - Latest master: 2022-12-09 03:32:47 +0000 (89fdafc27d) -- Refactor use of led_set_user within handwired/prime_exl (#19274)
Ψ - Latest upstream/master: 2022-12-10 21:35:06 -0500 (57351e4a37) -- [Keyboard] Add heliotrope Keyboard (#19170)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2022-12-09 03:32:47 +0000 (89fdafc27d) -- Refactor use of led_set_user within handwired/prime_exl (#19274)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-09-18 10:01:17 +0000 --  (0e9d558b5)
Ψ - lib/chibios-contrib: 2022-10-03 18:09:41 +0200 --  (bb8356fb)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

lalalademaxiya1, 2023-06-20:
@sajenim Hi, q4 has been merged into the qmk master branch. I apologize for not being able to reproduce your issue after compiling and flashing the code. Could you please try again now?

sajenim, 2023-06-30:
I believe the issue was created due to Gentoo GCC defaults as per #20525

closed 2023-06-30 by sajenim