qmk/qmk_firmware #19297: [Feature Request] Hold down Leader Key to hit the key sequence

oizehsgl, 2022-12-11:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

> English is not my native language; please excuse typing errors.
# what i want
1. hold KC_LEAD,tap KC_F,release KC_LEAD --> SEND_STRING("QMK is awesome.")
2. tap KC_LEAD --> tap_code(KC_ESC)
# why i need it
1. Sometimes your leader key is not on a comfortable place as the rest of keys on your sequence
2. #define LEADER_NO_TIMEOUT is a good solution
3. I choose another solution(move leader key to a comfortable place)
4. But there will be some problems
    - The comfortable place cannot be used effectively, because leader key is a single action key, can’t be used for anything else
    - Can't set LEADER_TIMEOUT to the right value for me
        1. type KC_LEAD ~ type KC_F = 300ms
        2. type KC_LEAD ~ type KC_N = 200ms
        3. so i defined LEADER_TIMEOUT = 300ms 
        4. I wasted 100ms when i type KC_LEAD,type KC_N
        5. if i defined LEADER_PER_KEY_TIMING,I will waste more time
# i tried
  1. Mod-Tap
  2. Tap-Dance
  3. change process_record_user()
  4. change leader_start() and leader_end()
But i didn't solve my problem,maybe I don't know enough about these functions.Finally i try to change the source code to solve my problem,it seems to work fine.
# my question
  1. Does anyone have the same need?
  2. Is there a solution without changing the source code?

sigprof, 2022-12-11:
You can call `qk_leader_start();` from a custom keycode handler — this will do the same thing as the `QK_LEAD` keycode. If you do that in `process_record_user()`, you should probably have `return false;` after that (if you return `true`, your custom keycode will also be a part of the leader key sequence — but you can also use that feature to have multiple leader keys with different behavior).

And you can either use a tap dance or [intercept a MT/LT keycode](https://docs.qmk.fm/#/mod_tap?id=intercepting-mod-taps) to implement different behavior for tap and hold.

oizehsgl, 2022-12-11:
> You can call `qk_leader_start();` from a custom keycode handler — this will do the same thing as the `QK_LEAD` keycode. If you do that in `process_record_user()`, you should probably have `return [...]

Thank you for your reply! I try it and qk_leader_start() work fine. 
here is my code
  1. config.h
    `#define LEADER_TIMEOUT 10000`
  2. keymap.c
```
LEADER_EXTERNS();
void matrix_scan_user(void) {
  LEADER_DICTIONARY() {
    leading = false;
    // demo
    SEQ_ONE_KEY(KC_F) {
      SEND_STRING("QMK is awesome.");
      did_leader_succeed = true;
    }
    leader_end();
  }
};
void leader_start(void) {
  did_leader_succeed = false;
}
void leader_end(void) {
  if (!did_leader_succeed) {
    tap_code(KC_ESC);
  }
}

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        case MY_LEAD_ESC:
            if(record->event.pressed) {
		qk_leader_start();
            	return false;
	    } else {
		leader_time=0;
            	return false;
	    }
    	    return true;
    }
    return true;
}
```

drashna, 2022-12-15:
> The above code has a bug. When i hold leader and tap another key six times,it will send ESC.

Your code says to send escape if the sequence did not succeed: 
```c
void leader_end(void) {
  if (!did_leader_succeed) {
    tap_code(KC_ESC);
  }
}
```

And it looks like leader keys only supports 5 keys in the sequence, so when you hit the sixth, it ends.

zzzachzzz, 2023-01-10:
Is there any way to pull this off, while still maintaining a low `LEADER_TIMEOUT`? The idea is that when the leader key is held the timeout would be bypassed, at least for the time between the leader key hold and the first key of the leader sequence, in the case that `LEADER_PER_KEY_TIMING` is set.

The use case I'm going after is binding Leader + j / k to Page Down / Page Up. Being a key useful for spamming, it would be ideal to be able to hold the leader key and then send multiple leader sequences while held.

Or if there's another feature that would be more appropriate to this use case let me know. I think of Leader as a custom modifier key.

closed 2023-05-11 by github-actions[bot]