qmk/qmk_firmware #19311: [Bug] Keychron Q6: F keys in macro not working

mrtnRitter, 2022-12-13:
### Describe the Bug

When a macro contains a F key, the macro is not send.

Steps to reproduce:
Fork this repo, compile using `qmk compile -kb keychron/q6/iso_encoder -km keychron`, open VIA, add macro {KC_LALT,KC_F4}, assign macro to a key, press the key and nothing happens.

Further investigation:
- keyboard was put into debug mode and QMK console prints all raw keypressed - when the macro was pressed, no keypress was registered (not 0x00, literally nothing as the key wasn't pressed)
- the issues persists for all keymaps
- the issues persists for Q6/ansi_encoder as well (see: https://www.reddit.com/r/olkb/comments/zkazsg/keychron_q6_no_f_keys_in_macros_bug_in_firmware/)
- downloaded QMK 0.18.17 and copy-replaced all files to latest fork of QMK, redo the steps above and macro did work as expected (see https://github.com/mrtnRitter/qmk_firmware.git)
- tested for keychron q6 encoder only, possible other keyboards are affected as well

### Keyboard Used

keychron/q6/iso_encoder

### Operating System

Windows 10

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: C:/Users/Martin/Documents/GitHub/qmk_firmware
Ψ Detected Windows 10 (10.0.19045).
Ψ Git branch: master
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-09-18 10:01:17 +0000 --  (0e9d558b5)
Ψ - lib/chibios-contrib: 2022-10-03 18:09:41 +0200 --  (bb8356fb)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

tzarc, 2022-12-13:
Likely a bug in VIA, not QMK.
As per the reply on your reddit post ([here](https://old.reddit.com/r/olkb/comments/zkazsg/keychron_q6_no_f_keys_in_macros_bug_in_firmware/j03iv3t/)) it seems VIA hasn't fully updated to the new keycode definitions within QMK.

Suggest posting an issue on their GitHub repo instead.

mrtnRitter, 2022-12-13:
Good idea making the macro in QMK to eliminate VIA. 
This is the first time creating a macro in keyboard.c - sorry in advanced if I made a beginners mistake. 
I get compiling errors. Send string is enabled in rules.mk

That's the code in keyboard.c below "// clang-format off" and above layer definition:
```
enum my_keycodes {
    QUIT = SAFE_RANGE,
};

bool process_record_keychron(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
    case QUIT:
        if (record->event.pressed) {
            SEND_STRING(X_F4);
        } 
        break;
    }
    return true;
};
```

That's the error:
```
./keyboards/keychron/q6/iso_encoder/keymaps/keychron/keymap.c: 
In function 'process_record_keychron':
quantum/send_string/send_string_keycodes.h:243:31: 
error: invalid suffix "d" on integer constant
  243 | #define X_F4                  3d
```

mrtnRitter, 2022-12-13:
Thanks to your help on Reddit, I can confirm that the macro works fine when implemented in QMK keyboard.c. 
This bug report is forwared to VIA. 

Thanks!

closed 2022-12-13 by mrtnRitter