qmk/qmk_firmware #19337: [Bug] Building manna-harbour_miryoku keymap on draculad fails

bashu, 2022-12-15:
### Describe the Bug

```sh
Ψ Compiling keymap with make --jobs=1 MIRYOKU_ALPHAS=DVORAK MIRYOKU_EXTRA=QWERTY MIRYOKU_CLIPBOARD=WIN draculad:manna-harbour_miryoku

QMK Firmware 0.19.3
Making draculad with keymap manna-harbour_miryoku and target clean

QMK Firmware 0.19.3
Making draculad with keymap manna-harbour_miryoku

avr-gcc (GCC) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_draculad/src/info_config.h                                                   [OK]
Generating: .build/obj_draculad/src/default_keyboard.c                                              [OK]
Generating: .build/obj_draculad/src/default_keyboard.h                                              [OK]
Compiling: keyboards/draculad/draculad.c                                                            [OK]
Compiling: .build/obj_draculad/src/default_keyboard.c                                               [OK]
Compiling: quantum/keymap_introspection.c                                                          In file included from quantum/keymap_introspection.c:9:
users/manna-harbour_miryoku/manna-harbour_miryoku.c: In function ‘u_td_fn_U_SYM’:
users/manna-harbour_miryoku/manna-harbour_miryoku.c:29:23: error: unsigned conversion from ‘int’ to ‘layer_state_t’ {aka ‘unsigned char’} changes value from ‘256’ to ‘0’ [-Werror=overflow]
   29 |     default_layer_set((layer_state_t)1 << U_##LAYER); \
      |                       ^~~~~~~~~~~~~~~~~~~~~~
users/manna-harbour_miryoku/miryoku_babel/miryoku_layer_list.h:20:1: note: in expansion of macro ‘MIRYOKU_X’
   20 | MIRYOKU_X(SYM,    "Sym") \
      | ^~~~~~~~~
users/manna-harbour_miryoku/manna-harbour_miryoku.c:32:1: note: in expansion of macro ‘MIRYOKU_LAYER_LIST’
   32 | MIRYOKU_LAYER_LIST
      | ^~~~~~~~~~~~~~~~~~
users/manna-harbour_miryoku/manna-harbour_miryoku.c: In function ‘u_td_fn_U_FUN’:
users/manna-harbour_miryoku/manna-harbour_miryoku.c:29:23: error: unsigned conversion from ‘int’ to ‘layer_state_t’ {aka ‘unsigned char’} changes value from ‘512’ to ‘0’ [-Werror=overflow]
   29 |     default_layer_set((layer_state_t)1 << U_##LAYER); \
      |                       ^~~~~~~~~~~~~~~~~~~~~~
users/manna-harbour_miryoku/miryoku_babel/miryoku_layer_list.h:21:1: note: in expansion of macro ‘MIRYOKU_X’
   21 | MIRYOKU_X(FUN,    "Fun")
      | ^~~~~~~~~
users/manna-harbour_miryoku/manna-harbour_miryoku.c:32:1: note: in expansion of macro ‘MIRYOKU_LAYER_LIST’
   32 | MIRYOKU_LAYER_LIST
      | ^~~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c: At top level:
quantum/keymap_introspection.c:20:1: error: static assertion failed: "Number of keymap layers exceeds maximum set by LAYER_STATE_(8|16|32)BIT"
   20 | _Static_assert(NUM_KEYMAP_LAYERS <= MAX_LAYER, "Number of keymap layers exceeds maximum set by LAYER_STATE_(8|16|32)BIT");
      | ^~~~~~~~~~~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make[1]: *** [.build/obj_draculad_manna-harbour_miryoku/quantum/keymap_introspection.o] Error 1
make: *** [draculad:manna-harbour_miryoku] Error 1
Make finished with errors
```

### Keyboard Used

draculad

### Operating System

macOS 12.6.1 21G217 x86_64

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /Users/bashu/Projects/qmk_firmware
Ψ Detected macOS 12.6.2 (Intel).
Ψ Git branch: master
Ψ Repo version: 0.19.3
Ψ - Latest master: 2022-12-15 08:09:14 +1100 (6543065180) -- Reorder paths to (hopefully) trigger a re-tag.
Ψ - Latest upstream/master: 2022-12-15 08:09:14 +1100 (6543065180) -- Reorder paths to (hopefully) trigger a re-tag.
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2022-12-15 08:09:14 +1100 (6543065180) -- Reorder paths to (hopefully) trigger a re-tag.
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 6.4
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-09-18 10:01:17 +0000 --  (0e9d558b5)
Ψ - lib/chibios-contrib: 2022-10-03 18:09:41 +0200 --  (bb8356fb)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [X] Karabiner (macOS)

fauxpark, 2022-12-15:
`draculad` has `#define LAYER_STATE_8BIT` in its config.h, most likely due to firmware size (it's a split board with rgblight, OLED, WPM, mousekey, NKRO and encoders enabled). So you can't have more than 8 layers on this board.

bashu, 2022-12-15:
to @fauxpark please check my PR

bashu, 2022-12-15:
> So you can't have more than 8 layers on this board.

Wrong, I can have more than 8 layers.

drashna, 2022-12-15:
just a heads up:

If via is enabled, then it defaults to 4 layers, max. You can change the max with the `DYNAMIC_KEYMAP_LAYER_COUNT`.

Also, if `DYNAMIC_KEYMAP_LAYER_COUNT` is 8 or less, it defaults to the 8 bit layer state. If it's between 8 and 16, then it defaults to the 16 bit layer state.

Additionally, the `LAYER_STATE_8BIT` define also limits the maximum number of layers to 8.  That's explicitly why you got the error.  The miryoku keymap uses more than 8 layers, so errors out because it's defined.

fauxpark, 2022-12-15:
To add to the above, it would probably be better to simply `#undef` it in the miryoku keymap's config.h, than move it to each individual keymap.

manna-harbour, 2022-12-15:
@fauxpark See https://github.com/manna-harbour/miryoku_qmk/pull/13.

bashu, 2022-12-15:
to @fauxpark done, PR updated

closed 2022-12-15 by drashna

manna-harbour, 2022-12-16:
>> To add to the above, it would probably be better to simply `#undef` it in the miryoku keymap's config.h, than move it to each individual keymap.

> @fauxpark See manna-harbour/miryoku_qmk#13.

@fauxpark It's a per-keymap setting. How is it better to define it at the keyboard level and undef in every other keymap, rather than defining it for the one keymap where it applies? Especially since it shouldn't have been added at the keyboard level in the first place, and isn't needed now at all anyway. See https://github.com/manna-harbour/miryoku_qmk/pull/13#issuecomment-1352523635.

PR #19338 (2022-12-15): Fix #19337

## Description

Moved LAYER_STATE_8BIT from `draculad/config.h` to `draculad/keymaps/default/config.h`, now I can build both default (with `VIA_ENABLE=yes`) and manna-harbour_miryoku keymaps.

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [x] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Closes #19337

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](http [... cut]