qmk/qmk_firmware #19340: [Bug] Print screen not working on macOS

ChaseHardin, 2022-12-15:
### Describe the Bug

I am creating a key mapping for the GMMK Pro keyboard and running into issues with print screen on a macOS. After reading the [documentation](https://docs.qmk.fm/#/keycodes), it appears `KC_PRINT_SCREEN` and `KC_PSCR` are supported on macOS, but for some reason the key doesn't do anything.

Are there any known workarounds for this? I've attempted some custom macros but run into weird edge cases. i.e. 

```c
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
    case CUSTOM_PRINT:
        if (record->event.pressed) {
            SEND_STRING(SS_DOWN(X_LCMD) SS_DOWN(X_LSFT) SS_DOWN(X_LCTL) SS_DOWN(X_4));
        } else {
        }
        break;
    }
    return true;
};
```

## Keymap configuration
```c
#include QMK_KEYBOARD_H

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {

//      ESC      F1       F2       F3       F4       F5       F6       F7       F8       F9       F10      F11      F12	     Prt           Rotary(Mute)
//      ~        1        2        3        4        5        6        7        8        9        0         -       (=)	     BackSpc           Del
//      Tab      Q        W        E        R        T        Y        U        I        O        P        [        ]        \                 PgUp
//      Caps     A        S        D        F        G        H        J        K        L        ;        "                 Enter             PgDn
//      Sh_L              Z        X        C        V        B        N        M        ,        .        ?                 Sh_R     Up       End
//      Ct_L     Win_L    Alt_L                               SPACE                               Alt_R    FN       Ct_R     Left     Down     Right

    [0] = LAYOUT(
        KC_ESC,  KC_F1,   KC_F2,   KC_F3,   KC_F4,   KC_F5,   KC_F6,   KC_F7,   KC_F8,   KC_F9,   KC_F10,  KC_F11,  KC_F12,  KC_PSCR, KC_MUTE,
        KC_GRV,  KC_1,    KC_2,    KC_3,    KC_4,    KC_5,    KC_6,    KC_7,    KC_8,    KC_9,    KC_0,    KC_MINS, KC_EQL,  KC_BSPC, KC_DEL,
        KC_TAB,  KC_Q,    KC_W,    KC_E,    KC_R,    KC_T,    KC_Y,    KC_U,    KC_I,    KC_O,    KC_P,    KC_LBRC, KC_RBRC, KC_BSLS, KC_PGUP,
        KC_CAPS, KC_A,    KC_S,    KC_D,    KC_F,    KC_G,    KC_H,    KC_J,    KC_K,    KC_L,    KC_SCLN, KC_QUOT,          KC_ENT,  KC_PGDN,
        KC_LSFT,          KC_Z,    KC_X,    KC_C,    KC_V,    KC_B,    KC_N,    KC_M,    KC_COMM, KC_DOT,  KC_SLSH,          KC_RSFT, KC_UP,   KC_END,
        KC_LCTL, KC_LGUI, KC_LALT,                            KC_SPC,                             KC_RALT, MO(1),   KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT
    ),

    [1] = LAYOUT(
        _______, KC_BRID, KC_BRIU, C(KC_UP), KC_F4, RGB_VAD, RGB_VAI, KC_MPRV, KC_MPLY, KC_MNXT, KC_MUTE, KC_VOLD, KC_VOLU, _______,          _______,
        _______, RGB_TOG, _______, _______, _______, _______, _______, _______, _______, _______, _______, _______, _______, _______,          _______,
        _______, _______, RGB_VAI, _______, _______, _______, _______, _______, _______, _______, _______, _______, _______, QK_BOOT,            _______,
        _______, _______, RGB_VAD, _______, _______, _______, _______, _______, _______, _______, _______, _______,          _______,          _______,
        _______,          _______, RGB_HUI, _______, _______, _______, NK_TOGG, _______, _______, _______, _______,          _______, RGB_MOD, _______,
        _______, _______, _______,                            _______,                            _______, _______, _______, RGB_SPD, RGB_RMOD, RGB_SPI
    ),
};
// clang-format on

#if defined(ENCODER_MAP_ENABLE)
const uint16_t PROGMEM encoder_map[][NUM_ENCODERS][2] = {
    [0] = { ENCODER_CCW_CW(KC_VOLD, KC_VOLU) },
    [1] = { ENCODER_CCW_CW(KC_TRNS, KC_TRNS) }
};
#endif

```

### Keyboard Used

gmmk/pro/rev1/ansi

### Operating System

macOS - Big Sur

### qmk doctor Output

```text
Making gmmk/pro/rev1/ansi with keymap hardin2

arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.07) 10.3.1 20210621 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   41478       0   41478    a206 gmmk_pro_rev1_ansi_hardin2.bin

Compiling: quantum/keymap_introspection.c                                                           [OK]
Linking: .build/gmmk_pro_rev1_ansi_hardin2.elf                                                      [OK]
Creating binary load file for flashing: .build/gmmk_pro_rev1_ansi_hardin2.bin                       [OK]
Creating load file for flashing: .build/gmmk_pro_rev1_ansi_hardin2.hex                              [OK]

Size after:
   text    data     bss     dec     hex filename
      0   41478       0   41478    a206 gmmk_pro_rev1_ansi_hardin2.bin

Copying gmmk_pro_rev1_ansi_hardin2.bin to qmk_firmware folder                                       [OK]
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

fauxpark, 2022-12-15:
Did you read the footnote?

ChaseHardin, 2022-12-15:
> Did you read the footnote?

@fauxpark I'm not sure what footnote you're referring to. Can you link me?

fauxpark, 2022-12-15:
The one for the print screen key on the docs page that you just linked to, in the macOS column.

ChaseHardin, 2022-12-15:
@fauxpark yes, I read the documentation and saw there is a checkbox which led me to believe this was supported functionality.

[image]

fauxpark, 2022-12-15:
And did you wonder what that little number next to it means?

ChaseHardin, 2022-12-15:
Yes, I wondered and spent an hour trying to find documentation on the differences. Can you please explain?

fauxpark, 2022-12-15:
Scroll down to the end of the table...

ChaseHardin, 2022-12-15:
Ok that is not obvious at a quick glance. I am going to create a PR to make the checkmark explanations more explicit. 

## Checkmark ✔️ definition
> 2. Treated as F13-F15.

This definition is a bit vague, can you help me understand how this applies in the print screen context?

fauxpark, 2022-12-15:
It's obvious to anyone who understands how footnotes work.

drashna, 2022-12-15:
Print Screen, Scroll lock, and Pause/break are treated as F13, F14, and F15, respectively. Not as the normal keycodes.

closed 2022-12-16 by ChaseHardin