qmk/qmk_firmware #19367: QK65 R2 Wired PCB Not Fully QMK Supported, Might Need Doc Change

ForsakenRei, 2022-12-17:
### Issue Description

Compared to QK65 R1 wired hotswap orange PCB, the R2 wired hotswap white PCB used a different bootloader(which is not supported by QMK). That bootloader takes a `.bin` firmware while QMK will compile a `.hex` firmware for QK65. Additionaly, QMK toolbox cannot recognize the bootloader.

That's to say end user cannot flash their own QMK compiled firmware on R2 PCB, even though it's QMK powered and Qwertykeys provided Via and Vial support. You can only use the firmware then provided.

https://github.com/qmk/qmk_firmware/tree/master/keyboards/qwertykeys/qk65/hotswap

The description here only works for the R1 PCB, so we probably need make it clear on that? I can open a pull request for the changes if necessary.

RivasMario, 2022-12-22:
Have they only attempted flashing with QMK? have they tried VIAL once again>

tzarc, 2022-12-22:
There's no R2 in QMK. Feel free to request the manufacturer rectify this fact.

closed 2022-12-22 by tzarc

ForsakenRei, 2022-12-22:
> There's no R2 in QMK. Feel free to request the manufacturer rectify this fact.

Yeah there is no R2 for QK65 in QMK, I guess basically because the reason above? Since it's not 'really' fully QMK supported, but QMK powered only.

I own the borad myself and talked to Syrah from Qwertykeys when I found I cannot flash my owner firmware. Per Syrah's word, the whole board and flash program was purchased as a canned solution so they don't know what happened to make a `.hex` firmware to be `.bin`. He said there was nothing else we can do. I raised this question in discord a while ago but at that time the message was soon buried and didn't really get meaningful response...

By rectifying the fact, do we need something official from Qwertykeys clarify that the R2 PCB cannot flash user compiled firmware?

tzarc, 2022-12-22:
If you want to be pedantic, _technically_ they can't distribute it with the binaries pre-flashed because the GPL requires source to be provided by the vendor.

I'd be trying to get them to push for this "canned solution" to be handed over from their contractors or whoever they used, then open-sourced, because if it's using QMK under the hood they're in breach of the license agreement with the current state of play.

ForsakenRei, 2022-12-23:
> If you want to be pedantic, _technically_ they can't distribute it with the binaries pre-flashed because the GPL requires source to be provided by the vendor.
>
> I'd be trying to get them to push for [...]

TBH I didn't really realize the GPL license part. And thanks for your great explanation, let me know if anything I can help.