qmk/qmk_firmware #19377: [Bug] Compilation fails for ferris/sweep with STM32F411

PotatoTech, 2022-12-18:
### Describe the Bug

Compilation fails when compiling firmware for `ferris/sweep` targeting STM32F411 with tinyuf2 bootloader. It also fails when using the Docker build script.
Tested on master and develop branches.

Output of `qmk compile -km default`:
<details>
<summary>Click to expand</summary>

```
Ψ Compiling keymap with make --jobs=1 ferris/sweep:default

QMK Firmware 0.19.4
Making ferris/sweep with keymap default

arm-none-eabi-gcc (Arch Repository) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/ferris/sweep/sweep.c                                                           [OK]
Compiling: .build/obj_ferris_sweep/src/default_keyboard.c                                           [OK]
Compiling: quantum/keymap_introspection.c                                                           [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/bitwise.c                                                                        [OK]
Compiling: quantum/led.c                                                                            [OK]
Compiling: quantum/action.c                                                                         [OK]
Compiling: quantum/action_layer.c                                                                   [OK]
Compiling: quantum/action_tapping.c                                                                 [OK]
Compiling: quantum/action_util.c                                                                    [OK]
Compiling: quantum/eeconfig.c                                                                       [OK]
Compiling: quantum/keyboard.c                                                                       [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/sync_timer.c                                                                     [OK]
Compiling: quantum/logging/debug.c                                                                  [OK]
Compiling: quantum/logging/sendchar.c                                                               [OK]
Compiling: quantum/logging/print.c                                                                  [OK]
Compiling: quantum/bootmagic/bootmagic_lite.c                                                       [OK]
Compiling: quantum/bootmagic/magic.c                                                                [OK]
Compiling: quantum/matrix_common.c                                                                  [OK]
Compiling: quantum/matrix.c                                                                        In file included from ./lib/chibios/os/hal/include/hal_pal.h:174,
                 from ./lib/chibios/os/hal/include/hal.h:299,
                 from ./platforms/chibios/drivers/wear_leveling/wear_leveling_efl_config.h:6,
                 from <command-line>:
./lib/chibios/os/hal/ports/STM32/LLD/GPIOv2/stm32_gpio.h:58:58: error: 'GPIOF_BASE' undeclared here (not in a function); did you mean 'GPIOH_BASE'?
   58 | #define GPIOF                           ((stm32_gpio_t *)GPIOF_BASE)
      |                                                          ^~~~~~~~~~
./lib/chibios/os/hal/ports/STM32/LLD/GPIOv2/hal_pal_lld.h:171:26: note: in definition of macro 'PAL_LINE'
  171 |   ((ioline_t)((uint32_t)(port)) | ((uint32_t)(pad)))
      |                          ^~~~
platforms/chibios/_pin_defs.h:201:21: note: in expansion of macro 'GPIOF'
  201 | #define F7 PAL_LINE(GPIOF, 7)
      |                     ^~~~~
./keyboards/ferris/sweep/config.h:29:11: note: in expansion of macro 'F7'
   29 |     { E6, F7, F6, F5, F4 }, \
      |           ^~
quantum/matrix.c:54:70: note: in expansion of macro 'DIRECT_PINS'
   54 | static SPLIT_MUTABLE pin_t direct_pins[ROWS_PER_HAND][MATRIX_COLS] = DIRECT_PINS;
      |                                                                      ^~~~~~~~~~~
 [ERRORS]
 |
 |
 |
make[1]: *** [builddefs/common_rules.mk:360: .build/obj_ferris_sweep_default/quantum/matrix.o] Error 1
Make finished with errors
make: *** [Makefile:415: ferris/sweep:default] Error 1
```

</details>

Contents of `keyboards/ferris/sweep/rules.mk`:
<details>
<summary>Click to expand</summary>

```
# MCU name
MCU = STM32F411

# Bootloader selection
BOOTLOADER = tinyuf2

# Build Options
#   change yes to no to disable
#
BOOTMAGIC_ENABLE = yes      # Enable Bootmagic Lite
MOUSEKEY_ENABLE = yes       # Mouse keys
EXTRAKEY_ENABLE = yes       # Audio control and System control
CONSOLE_ENABLE = no         # Console for debug
COMMAND_ENABLE = no         # Commands for debug and configuration
NKRO_ENABLE = no            # Enable N-Key Rollover
BACKLIGHT_ENABLE = no       # Enable keyboard backlight functionality
RGBLIGHT_ENABLE = no        # Enable keyboard RGB underglow
UNICODE_ENABLE = yes        # Unicode
AUDIO_ENABLE = no           # Audio output
SPLIT_KEYBOARD = yes        # Use shared split_common code
LAYOUTS = split_3x5_2
```

</details>

All other files are unmodified.

### Keyboard Used

ferris/sweep

### Operating System

Arch Linux

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /home/<redacted>/qmk_firmware
Ψ Detected Linux.
Ψ Git branch: develop
Ψ Repo version: 0.19.4
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest develop: 2022-12-18 21:55:14 +0100 (c2b13bd77b) -- Introduce VERIFY_AND_CLEAR shorthand (#19370)
Ψ - Latest upstream/master: 2022-12-15 11:35:28 -0800 (f21443d6a2) -- DZTECH OG60 Touch-Up (#19343)
Ψ - Latest upstream/develop: 2022-12-18 21:55:14 +0100 (c2b13bd77b) -- Introduce VERIFY_AND_CLEAR shorthand (#19370)
Ψ - Common ancestor with upstream/master: 2022-12-15 11:35:28 -0800 (f21443d6a2) -- DZTECH OG60 Touch-Up (#19343)
Ψ - Common ancestor with upstream/develop: 2022-12-18 21:55:14 +0100 (c2b13bd77b) -- Introduce VERIFY_AND_CLEAR shorthand (#19370)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-09-18 10:01:17 +0000 --  (0e9d558b5)
Ψ - lib/chibios-contrib: 2022-12-09 23:26:27 +0100 --  (1130173e)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

tzarc, 2022-12-18:
You're going to need to explain what you're trying to do -- `ferris/sweep` targets 32U4.

PotatoTech, 2022-12-18:
I am trying to use the STeMCell microcontrollers [(docs)](https://megamind4089.github.io/STeMCell/) which is a Pro Micro / Elite C replacement. These use STM32F411 chips.

zvecr, 2022-12-18:
`qmk compile -kb ferris/sweep -km default -e CONVERT_TO=stemcell` as a guess?

Randomly changing rules.mk without updating the remaining config isnt the way to go...

PotatoTech, 2022-12-18:
Issue resolved. I read the STeMCell portion of the converters page but missed the `CONVERT_TO` command argument at the top. Thanks for the quick response.

closed 2022-12-18 by PotatoTech