qmk/qmk_firmware #19381: [Bug] Combo which produces `KC_ESC` does not break out of Caps Word

adiron, 2022-12-19:
### Describe the Bug

I have the combo of J+K set up to perform `KC_ESC`. The characters J and K are of course not stop keys for Caps Word.

Inputting the following:
`<Both Shifts to enter Caps Word>hello<J+K combo to perform Escape>world`
will produce HELLOWORLD

Whereas:
`<Both Shifts to enter Caps Word>hello<a key straight up bound to KC_ESC>world`
will produce HELLOworld (as expected)

I suspect that Caps Word does not recognize J+K as stop chars since both `KC_J` and `KC_K` are acceptable, and does not check for the fact that `KC_ESC` was fired.

### Keyboard Used

xiudi/xd75:adi

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

Exact keymap to reproduce this keyboard exists in my own fork of qmk_firmware

sigprof, 2022-12-19:
This happens because you have `[COMBO_ESC] = COMBO_ACTION(combo_esc)` and use `process_combo_event()` to implement the combo as `tap_code(KC_ESC)`.  Any direct `tap_code()` calls are not processed by Caps Word or any other event processing code.

You should use `[COMBO_ESC] = COMBO(combo_esc, KC_ESC)` instead and drop the handling in `process_combo_event()`; in this case the resulting `KC_ESC` will be processed by `process_record_user()` and any other event processing functions, including the Caps Word feature.

BTW, using `process_combo_event()` may be problematic if you also have dual-role keys like `LT()` or `MT()` in your keymap, because `process_combo_event()` may be called too early in some cases (if you press a combo quickly after pressing a dual-role key, `process_combo_event()` may be called before the dual-role key is actually handled).  To avoid this problem, you should map the combo to a custom keycode (combos now support any keycodes as their outputs including even things like `LT()` or `MT()`), and then handle that keycode in `process_record_user()` as if it was on a real key; then `process_combo_event()` could be removed completely.

adiron, 2022-12-20:
Oh, didn't know that. Thank you for the advice, very solid stuff!

Maybe this pitfall should be stated in the docs, but I leave that up to someone more knowledgeable than me

Thanks for he help!

closed 2022-12-20 by adiron