qmk/qmk_firmware #19401: Recommend the use of compilation database for VS Code

wbkang, 2022-12-22:
### Issue Description

The current documentation recommends a VS code setup that's not quite right. VS Code is capable of understanding the standard compile_commands.json database, so we should recommend that as the default setup. I've noticed that https://github.com/qmk/qmk_firmware/issues/15434 was closed but the documentation for vscode wasn't updated so I am opening this issue to track that.

flowchartsman, 2023-02-05:
Is there anything else that can be added to these instructions? I found them while trying to get intellisense working correctly on my M1 macbook to do some work on an RGB matrix animation, but I did not run across an `-mmcu` flag in `compile_commands.json`, and I am also getting a message that says:
```
"rgb_matrix_effects.inc" not found in "${workspaceFolder}/compile_commands.json". 'includePath' from c_cpp_properties.json in folder 'qmk' will be used for this file instead.
```

And I'm getting random and inconsistent
```
#include errors detected. Consider updating your compile_commands.json or includePath. Squiggles are disabled for this translation unit (/Users/<me>/devel/qmk/keyboards/keychron/q6/ansi_encoder/ansi_encoder.c).C/C++(1696)
```
Which isn't particularly clear, and it's not clear what defines I need, since building with the `qmk` cli results in a working firmware that has most, if not all, RGB effects/animations enabled, so maybe `generate-compilation-database` is inconsistent somehow.

The troubleshooting instructions are also a bit vague. It's not entirely clear what "not matching what you expect" is supposed to mean, and they state "if it's not part of your compilation database, it will tell you so", but I don't get any messages about anything missing, nor do I see any defines listed in this output, despite having set them in both the config.h for the keyboard as well as in `rules.mk`.

For example, I am starting off by modifying an existing file `quantum/rgb_matrix/animations/pixel_fractal.h`, which seems to require two defines: `ENABLE_RGB_MATRIX_PIXEL_FRACTAL` and `RGB_MATRIX_CUSTOM_EFFECT_IMPLS`, both of which are defined in the config.h for my keyboard, which is in `keyboards/keychron/q6/ansi_encoder/config.h`, yet the code is greyed out,  and on accessing the file, it reports include errors on the first character of the file.

flowchartsman, 2023-02-05:
For completness' sake, the output of my C/C++ diagnostics is:
```
-------- Diagnostics - 2/4/2023, 8:29:44 PM
Version: 1.13.9
Current Configuration:
{
    "name": "qmk",
    "compilerArgs": [],
    "compilerPath": "/opt/homebrew/bin/arm-none-eabi-gcc",
    "cStandard": "c11",
    "cppStandard": "c++17",
    "compileCommands": "${workspaceFolder}/compile_commands.json",
    "intelliSenseMode": "linux-gcc-arm",
    "browse": {
        "path": [
            "${workspaceFolder}"
        ],
        "limitSymbolsToIncludedHeaders": true
    },
    "compilerPathIsExplicit": true,
    "cStandardIsExplicit": true,
    "cppStandardIsExplicit": true,
    "intelliSenseModeIsExplicit": true,
    "mergeConfigurations": false
}
Translation Unit Mappings:
[ /my/path/to/qmk/quantum/rgb_matrix/animations/rgb_matrix_effects.inc ]:
    /my/path/to/qmk/quantum/rgb_matrix/animations/pixel_flame_anim.h *
Translation Unit Configurations:
[ /my/path/to/qmk/quantum/rgb_matrix/animations/rgb_matrix_effects.inc ]:
    Process ID: 63001
    Memory Usage: 7 MB
    Compiler Path: /opt/homebrew/bin/arm-none-eabi-gcc
    Includes:
        /opt/homebrew/Cellar/arm-gcc-bin@8/8-2019-q3-update_2/lib/gcc/arm-none-eabi/8.3.1/include
        /opt/homebrew/Cellar/arm-gcc-bin@8/8-2019-q3-update_2/lib/gcc/arm-none-eabi/8.3.1/include-fixed
        /opt/homebrew/Cellar/arm-gcc-bin@8/8-2019-q3-update_2/arm-none-eabi/include
    Standard Version: c11
    IntelliSense Mode: linux-gcc-arm
    Other Flags:
        --gcc
        --gnu_version=80301
Total Memory Usage: 7 MB
Browse Paths from compile_commands.json, from workspace folder: /my/path/to/qmk
    ./platforms/chibios/boards/GENERIC_STM32_L432XC/configs
    ./tmk_core/protocol/chibios
    /my/path/to/qmk
    /my/path/to/qmk/.
    /my/path/to/qmk/./lib/chibios/os/common/ext/ARM/CMSIS/Core/Include
    /my/path/to/qmk/./lib/chibios/os/common/ext/ST/STM32L4xx
    /my/path/to/qmk/./lib/chibios/os/common/portability/GCC
    /my/path/to/qmk/./lib/chibios/os/common/ports/ARM-common
    /my/path/to/qmk/./lib/chibios/os/common/ports/ARMv7-M
    /my/path/to/qmk/./lib/chibios/os/common/startup/ARMCMx/compilers/GCC
    /my/path/to/qmk/./lib/chibios/os/common/startup/ARMCMx/devices/STM32L4xx
    /my/path/to/qmk/./lib/chibios/os/hal/boards/ST_NUCLEO32_L432KC
    /my/path/to/qmk/./lib/chibios/os/hal/include
    /my/path/to/qmk/./lib/chibios/os/hal/lib/streams
    /my/path/to/qmk/./lib/chibios/os/hal/osal/rt-nil
    /my/path/to/qmk/./lib/chibios/os/hal/ports/STM32/LLD/ADCv3
    /my/path/to/qmk/./lib/chibios/os/hal/ports/STM32/LLD/CANv1
    /my/path/to/qmk/./lib/chibios/os/hal/ports/STM32/LLD/DACv1
    /my/path/to/qmk/./lib/chibios/os/hal/ports/STM32/LLD/DMAv1
    /my/path/to/qmk/./lib/chibios/os/hal/ports/STM32/LLD/EXTIv1
    /my/path/to/qmk/./lib/chibios/os/hal/ports/STM32/LLD/GPIOv3
    /my/path/to/qmk/./lib/chibios/os/hal/ports/STM32/LLD/I2Cv2
    /my/path/to/qmk/./lib/chibios/os/hal/ports/STM32/LLD/QUADSPIv1
    /my/path/to/qmk/./lib/chibios/os/hal/ports/STM32/LLD/RCCv1
    /my/path/to/qmk/./lib/chibios/os/hal/ports/STM32/LLD/RNGv1
    /my/path/to/qmk/./lib/chibios/os/hal/ports/STM32/LLD/RTCv2
[... 82 lines cut ...]
    lib/chibios/os/oslib/src
    lib/chibios/os/rt/src
    lib/chibios/os/various
    lib/fnv
    lib/lib8tion
    lib/printf/src/printf
    platforms
    platforms/chibios
    platforms/chibios/bootloaders
    platforms/chibios/drivers
    platforms/chibios/drivers/wear_leveling
    quantum
    quantum/bootmagic
    quantum/debounce
    quantum/logging
    quantum/process_keycode
    quantum/rgb_matrix
    quantum/send_string
    quantum/wear_leveling
    tmk_core/protocol
    tmk_core/protocol/chibios

------- Workspace parsing diagnostics -------
Number of files discovered (not excluded): 56098
```

wbkang, 2023-02-05:
Hey @flowchartsman, this is a pull request to improve the documentation RE: the use of compilation database and I think your issue is probably not related. I did look at your issue and I think I kind of know why:

```
"rgb_matrix_effects.inc" not found in "${workspaceFolder}/compile_commands.json"
```

The compilation database (or any C/C++ auto-completes for that matter) work against a [compilation unit](https://stackoverflow.com/a/1106167/2710) - which is usually a c/c++ file that is compiled and then results in an object file. The error message is given because `rgb_matrix_effects.inc` is not one of the files that are being compiled (it's probably included as part of another file). Auto-completes for header files are inherently limited and heuristics based because they lack the context in which the compilation happens.

```
#include errors detected. Consider updating your compile_commands.json or includePath. Squiggles are disabled for this translation unit (/Users/<me>/devel/qmk/keyboards/keychron/q6/ansi_encoder/ansi_encoder.c)
```

Can you check your compile_commands.json? It's a fairly simple JSON, you should be able to find a list of source code being compiled. Do you see `keyboards/keychron/q6/ansi_encoder/ansi_encoder.c` in the file? If not, what do you see? Make sure to pass in the right keyboard & keymap when running  `qmk generate-compilation-database -kb <keyboard> -km <keymap>`

flowchartsman, 2023-02-05:
> Can you check your compile_commands.json? It's a fairly simple JSON, you should be able to find a list of source code being compiled. Do you see keyboards/keychron/q6/ansi_encoder/ansi_encoder.c in [...]

Thank you for responding! And yes, I definitely checked it first. I have set the keyboard as default with:
```
qmk config user.keyboard=keychron/q6/ansi_encoder
qmk config user.keymap=flowchartsman
```

But I have also generated the json file explicitly with

```
qmk generate-compilation-database -kb keychron/q6/ansi_encoder -km flowchartsman
```

And the result is the same. Include errors all over the place.

flowchartsman, 2023-02-05:
Not to mention this cryptic error, which I assume is some kind of syntax issue caused by the `#define`s not being correct. But it's not particularly clear how to proceed
[image]

flowchartsman, 2023-02-05:
This is why I mention it here. Because I followed the steps, which make sense, and yet VSCode is still behaving unpredictably, so either something is missing from the instructions, or there's an alternate path that I should be taking that is not documented (or it is unclear where to look from the docs)

wbkang, 2023-02-05:
Do you mind checking if you specifically see `keyboards/keychron/q6/ansi_encoder/ansi_encoder.c` in `compile_commands.json`? And if you see it, what does the command say exactly? Because the error message `Consider updating your compile_commands.json or includePath. Squiggles are disabled for this translation unit (/Users/<me>/devel/qmk/keyboards/keychron/q6/ansi_encoder/ansi_encoder.c).C/C++(1696)` is specifically saying ansi_encoder.c file is not found in your compile_commands.json.

I wouldn't pay attention to the errors that are not in your current open C file. vscode c/cpp tool only operates on individual C files only, and other errors can be inaccurate.

RE: the error "expected an identifier" it simply (and vaguely) means it expects an identifier (e.g., a name of a variable) to be followed where it is. 

For example, given the following file
```
% cat test.c
int () {
}
```

gcc will complain 
```
test.c:1:6: error: expected identifier or ‘(’ before ‘)’ token
    1 | int () {
      |      ^
```

Because it expects a function name (an identifier) at line 1, column 6 - but instead there is something else that is not an identifier. The error is necessarily cryptic because the IDE does not have all the right defines, and therefore it's probably parsing a non-syntactical header file to begin with.

---

To rephrase my original points: `compile_commands.json` will not really help much with any **non-C** file, such as headers. If you jump to the header from a C file that is IN `compile_commands.json`, the IDE can guess in which C file is this header included and so it can understand the header. But this is heuristics-based rather than something that always works reliably.

If you still want it to work, then you have to extract the right defines, include paths from compile_commands.json for one of the source code and [configure `c_cpp_properties.json` manually](https://code.visualstudio.com/docs/cpp/c-cpp-properties-schema-reference). This is imperfect because compile_commands.json allows defines/include files to be correctly specified for each C file, whereas the c_cpp_properties.json can only specify them project-wide globally, but it works well enough.

flowchartsman, 2023-02-05:
Thanks for the explanation. I was under the impression that the defines were extracted from `compile_commands.json`, since the extension seems to indicate this in the advanced settings section:
[image]

Which is what lead me to wonder if somehow `compile_commands.json` was incomplete.

As for `ansi_encoder.c`, it does appear:

```
    {
        "directory": "/Users/<me>/devel/qmk",
        "command": "arm-none-eabi-gcc -c -mcpu=cortex-m4 -mthumb -DTHUMB_PRESENT -DTHUMB_NO_INTERWORKING -mno-unaligned-access -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -fomit-frame-pointer -ffunction-sections -fdata-sections -fno-common -fshort-wchar -fno-builtin-printf -ggdb -flto -DLTO_ENABLE -Os -Wall -Wstrict-prototypes -Werror -std=gnu11 -fcommon -DCORTEX_ENABLE_WFI_IDLE=TRUE -DKEYBOARD_ -DKEYBOARD_ -DKEYBOARD_keychron -DKEYBOARD_keychron_q6 -DKEYBOARD_keychron_q6_ansi_encoder -DMCU_STM32 '-DKEYMAP_C=\"keyboards/keychron/q6/ansi_encoder/keymaps/flowchartsman/keymap.c\"' -DPRINTF_SUPPORT_DECIMAL_SPECIFIERS=0 -DPRINTF_SUPPORT_EXPONENTIAL_SPECIFIERS=0 -DPRINTF_SUPPORT_LONG_LONG=0 -DPRINTF_SUPPORT_WRITEBACK_SPECIFIER=0 -DSUPPORT_MSVC_STYLE_INTEGER_SPECIFIERS=0 -DPRINTF_ALIAS_STANDARD_FUNCTION_NAMES=1 -DMOUSEKEY_ENABLE -DEEPROM_ENABLE -DEEPROM_DRIVER -DEEPROM_WEAR_LEVELING -DWEAR_LEVELING_ENABLE -DWEAR_LEVELING_EMBEDDED_FLASH -DHAL_USE_EFL -DRGB_MATRIX_ENABLE -DCKLED2001 -DSTM32_I2C -DHAL_USE_I2C=TRUE -DUSE_CIE1931_CURVE -DBOOTMAGIC_LITE -DFNV_ENABLE -DMAGIC_KEYCODE_ENABLE -DSEND_STRING_ENABLE -DENCODER_ENABLE -DENCODER_MAP_ENABLE -DDIP_SWITCH_ENABLE -DGRAVE_ESC_ENABLE -DSPACE_CADET_ENABLE -DMOUSE_ENABLE -DMOUSE_SHARED_EP -DEXTRAKEY_ENABLE -DNO_PRINT -DNO_DEBUG -DNKRO_ENABLE -DSHARED_EP_ENABLE -DBOOTLOADER_STM32_DFU -DSTM32_BOOTLOADER_ADDRESS=0x1FFF0000 -DCORTEX_USE_FPU=TRUE -DPROTOCOL_CHIBIOS -DPLATFORM_SUPPORTS_SYNCHRONIZATION -DPORT_IGNORE_GCC_VERSION_CHECK=1 -DFIXED_CONTROL_ENDPOINT_SIZE=64 -DFIXED_NUM_CONFIGURATIONS=1 -DQMK_MCU=STM32L432 -DQMK_MCU_STM32L432 -DQMK_MCU_ARCH=cortex-m4 -DQMK_MCU_ARCH_CORTEX_M4 -DQMK_MCU_PORT_NAME=STM32 -DQMK_MCU_PORT_NAME_STM32 -DQMK_MCU_FAMILY=STM32 -DQMK_MCU_FAMILY_STM32 -DQMK_MCU_SERIES=STM32L4xx -DQMK_MCU_SERIES_STM32L4XX -DQMK_BOARD=GENERIC_STM32_L432XC -DQMK_BOARD_GENERIC_STM32_L432XC '-DQMK_KEYBOARD=\"keychron/q6/ansi_encoder\"' '-DQMK_KEYBOARD_H=\".build/obj_keychron_q6_ansi_encoder/src/default_keyboard.h\"' '-DQMK_KEYMAP=\"flowchartsman\"' '-DQMK_KEYMAP_H=\"flowchartsman.h\"' '-DQMK_KEYMAP_CONFIG_H=\"keyboards/keychron/q6/ansi_encoder/keymaps/flowchartsman/config.h\"' -Ikeyboards/keychron/q6/ansi_encoder/keymaps/flowchartsman -Iusers/flowchartsman -Ikeyboards/. -Ikeyboards/. -Ikeyboards/keychron -Ikeyboards/keychron/q6 -Ikeyboards/keychron/q6/ansi_encoder -I./platforms/chibios/boards/GENERIC_STM32_L432XC/configs -I. -Itmk_core -Iquantum -Iquantum/keymap_extras -Iquantum/process_keycode -Iquantum/sequencer -Idrivers -Iplatforms/chibios/drivers/eeprom -Idrivers/eeprom -Iplatforms/chibios/drivers/wear_leveling -Idrivers/wear_leveling -Iquantum/wear_leveling -Iquantum/rgb_matrix -Iquantum/rgb_matrix/animations -Iquantum/rgb_matrix/animations/runners -Idrivers/led -Iquantum/bootmagic -Iquantum/send_string -I.build/obj_keychron_q6_ansi_encoder/src -I.build/obj_keychron_q6_ansi_encoder_flowchartsman/src -Iquantum/logging -Ilib/printf/src -Ilib/printf/src/printf -Ilib/fnv -Itmk_core/protocol -Iplatforms -Iplatforms/chibios -Iplatforms/chibios/drivers -Itmk_core/protocol -Itmk_core/protocol/chibios -Itmk_core/protocol/chibios/lufa_utils -I./lib/chibios/os/license -I./lib/chibios/os/oslib/include -I./platforms/chibios/boards/GENERIC_STM32_L432XC/configs -I./platforms/chibios/boards/common/configs -Ikeyboards/keychron/q6 -I./lib/chibios/os/common/portability/GCC -I./lib/chibios/os/common/startup/ARMCMx/compilers/GCC -I./lib/chibios/os/common/startup/ARMCMx/devices/STM32L4xx -I./lib/chibios/os/common/ext/ARM/CMSIS/Core/Include -I./lib/chibios/os/common/ext/ST/STM32L4xx -I./lib/chibios/os/rt/include -I./lib/chibios/os/common/portability/GCC -I./lib/chibios/os/common/ports/ARM-common -I./lib/chibios/os/common/ports/ARMv7-M -I./lib/chibios/os/hal/osal/rt-nil -I./lib/chibios/os/hal/include -I./lib/chibios/os/hal/ports/common/ARMCMx -I./lib/chibios/os/hal/ports/STM32/STM32L4xx -I./lib/chibios/os/hal/ports/STM32/LLD/ADCv3 -I./lib/chibios/os/hal/ports/STM32/LLD/CANv1 -I./lib/chibios/os/hal/ports/STM32/LLD/DACv1 -I./lib/chibios/os/hal/ports/STM32/LLD/DMAv1 -I./lib/chibios/os/hal/ports/STM32/LLD/EXTIv1 -I./lib/chibios/os/hal/ports/STM32/LLD/GPIOv3 -I./lib/chibios/os/hal/ports/STM32/LLD/I2Cv2 -I./lib/chibios/os/hal/ports/STM32/LLD/QUADSPIv1 -I./lib/chibios/os/hal/ports/STM32/LLD/RCCv1 -I./lib/chibios/os/hal/ports/STM32/LLD/RNGv1 -I./lib/chibios/os/hal/ports/STM32/LLD/RTCv2 -I./lib/chibios/os/hal/ports/STM32/LLD/SDMMCv1 -I./lib/chibios/os/hal/ports/STM32/LLD/SPIv2 -I./lib/chibios/os/hal/ports/STM32/LLD/SYSTICKv1 -I./lib/chibios/os/hal/ports/STM32/LLD/TIMv1 -I./lib/chibios/os/hal/ports/STM32/LLD/USART -I./lib/chibios/os/hal/ports/STM32/LLD/USARTv2 -I./lib/chibios/os/hal/ports/STM32/LLD/USBv1 -I./lib/chibios/os/hal/ports/STM32/LLD/xWDGv1 -I./lib/chibios/os/hal/boards/ST_NUCLEO32_L432KC -I./lib/chibios/os/hal/lib/streams -I./lib/chibios/os/various -I. -Itmk_core -Iquantum -Iquantum/keymap_extras -Iquantum/process_keycode -Iquantum/sequencer -Idrivers -Iplatforms/chibios/drivers/eeprom -Idrivers/eeprom -Iplatforms/chibios/drivers/wear_leveling -Idrivers/wear_leveling -Iquantum/wear [... cut ...]
To your point about extracting the defines, I actually tried this already. With a little vim regexp magic on `compile_commands.json`, and  a pipe through `| sort |uniq` I added those to `c_cpp_properties.json`, but it didn't appear to make a difference. Would I then be disabling the `compile_commands.json` parsing after doing that? I'm not particularly clear on that, either.

I understand the heuristic difficulties, certainly. But it seems as if people are developing on the project successfully with VS Code, so it's unclear why I'm having so many issues.

wbkang, 2023-02-05:
> Thanks for the explanation. I was under the impression that the defines were extracted from compile_commands.json, since the extension seems to indicate this in the advanced settings section:

Yes, this is correct. If the file is found in compile_commands.json, vscode seems to ignore the other configs entirely (found out just by trying out things). If the file is NOT found, then vscode uses the defines from that config file.

I don't have an M1 mac, but I did check it out under Linux (WSL) using ` -kb keychron/q6/ansi_encoder -km default` and ansi_encoder.c doesn't seem to have any issues on my computer. For your reference - [this is my diagnostics dump](https://pastebin.com/CDFyuAmF). You can compare it against your diagnostics log to see how it is different, while you have `keyboards/keychron/q6/ansi_encoder/ansi_encoder.c` file open. 

[image]

RE: the files you originally reported as not working (e.g., rgb_matrix_effects.inc) 

I did look at it, the heuristics are correct because rgb_matrix.c does include them. I do see the same errors. There's a lot of macro magic involved though, so my guess is that either the compile_commands.json is not sufficient for vscode to get all the right defines (that's why I included that `-mcu` thing for my setup to work around this bug), or there is a bug with the macro expansion. I am not too familiar with it, so I am not sure if I can help you with this. 😢  Personally, I'd just disable the squiggles and move on. You _can_ try generating the preprocesed C file by using `gcc -E` but I couldn't find how to hack the makefile to do this.

If you build inside the vscode terminal, you can still jump to problems from the build output - so that's good enough for me.

joshuataylor, 2023-04-09:
I couldn't get `generate-compilation-database` to work with either VSCode or CLion. I just want nice autocomplete, not actual building.

If you are using a Mac, replace `make` with `gmake`, which will use GNU Make (if you installed qmk via Homebrew).

What did make this work was the following:

1. Install [bear](https://github.com/rizsotto/Bear), which can generate the `compile_commands.json` for us. It wraps around make.
2. Run `bear make --jobs=1 KEYBOARD_LOCATION:KEYMAP`, for example my keyboard location is `keychron/v6/ansi_encoder` and keymap is `joshuataylor`. If you're stuck, grab the command out of `qmk compile`.
3. Your `compile_commands.json` should have everything in there.

CLion worked fine with this, open as a "Compilation database project", not a "Makefile project".

VSCode however needed to have the paths added, as it didn't seem to read the `compile_commands.json` properly, I needed to add the paths.

I think bear gets stuck when it's wrapped around `qmk compile`, as it's a wrapper of a wrapper :)

eliliam, 2023-06-02:
@joshuataylor which paths did you need to update in VSCode? I'm having a challenging time getting all of these errors resolved on an Apple Silicon mac. `qmk compile` works fine, but I still have warnings and errors all over when I open the `keymap.c` file. Can you please also confirm that these assumptions of mine are correct? 
1. The `compile_commands.json` file lives in the root of the `qmk_firmware` project
2. VSCode needs to open the `qmk_firmware` folder as it's root
3. When running `bear` it should be ran in the root of `qmk_firmware` and not within a subdirectory(such as my specific keymap I am working on)
4. `qmk` commands can be ran wherever, does not have to be in the root of the project.

closed 2023-10-01 by github-actions[bot]