qmk/qmk_firmware #19410: [Feature Request] Add Keychron K8 Pro to the configurator (ISO)

CodusMaximus, 2022-12-23:
### Feature Request Type

- [ ] Core functionality
- [X] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

### Description

Hi there,

I have a Keychron K8 Pro ISO, It's not available on the QMK configurator.

Is this the appropriate place to request it to be added on?

Is there an alternative method I can use to customize the keyboard using something like this:
https://github.com/Keychron/qmk_firmware/tree/bluetooth_playground/keyboards/keychron/k8_pro

Many thanks in advance,

waffle87, 2022-12-23:
An alternative GUI configuration tool would be [VIA](https://caniusevia.com) — ask in Keychron Discord server for assistance using it with this keyboard. It is up to Keychron to open a PR for this keyboard to supported in upstream QMK/the Configurator.

CodusMaximus, 2022-12-23:
> An alternative GUI configuration tool would be [VIA](https://caniusevia.com) — ask in Keychron Discord server for assistance using it with this keyboard. It is up to Keychron to open a PR for this keyboard to supported in upstream QMK/the Configurator. 

Thanks for the quick reply,

VIA doesn't support the level of customisation I'm looking for unfortunately 

As for an alternative,
I was thinking more along the lines of forking their most stable version and editing/compiling myself

Am I right in assuming that the configurator (for the most part) edits the config files, builds and the toolbox deploys?

I.e. if I edit the code/configs manually and compile - can I still use the toolbox to deploy to the keyboard?

Thanks again,

waffle87, 2022-12-23:
> VIA doesn't support the level of customisation I'm looking for unfortunately

Then the Configurator doesn't either. 

And yes, that's a perfectly reasonable alternative. QMK Toolbox is pretty much just a flashing tool, so even that isn't necessarily required. You can use [`qmk flash`](https://docs.qmk.fm/#/newbs_flashing?id=flash-your-keyboard-from-the-command-line) to compile and flash the firmware all in one go.

CodusMaximus, 2022-12-23:
> Then the Configurator doesn't either.
>
> And yes, that's a perfectly reasonable alternative. QMK Toolbox is pretty much just a flashing tool, so even that isn't necessarily required. You can use [...]

Oh, I was hoping to use some of the leader key and dynamic macros from the QMK docs,

They don't seem to be available using VIA

Does that mean they haven't implemented those then?

waffle87, 2022-12-23:
VIA allows for basic key remapping, lighting control, and macros. It doesn't expose really any advanced QMK features. Plan on configuring those things directly in QMK source code.

CodusMaximus, 2022-12-23:
Understood, thanks for your support on this Jack 🙂

In that case, I need to try and get access to Keychron's version if they'll allow it - I don't want to risk bricking the keyboard

Feel free to close this, I appreciate the guidance

Merry Christmas!

closed 2023-02-02 by tzarc

TimSuss, 2023-11-16:
> It is up to Keychron to open a PR for this keyboard to supported in upstream QMK/the Configurator.

Could someone briefly ELI5 how I could facilitate this PR that @waffle87 suggests? Would I submit an issue to the Keychron github repo and ask someone to submit a PR to this repo?

adophoxia, 2023-11-16:
@TimSuss
>  Would I submit an issue to the Keychron github repo and ask someone to submit a PR to this qmk_firmware repo?

I wouldn't bother. The code that's needed to make BT work for QMK for Keychron's boards is  too custom in its state. For that to change, they'd have to heavily modify their code to be aligned with current QMK standards, which I doubt will change in the future.