qmk/qmk_firmware #19425: [Feature Request] Evision VS11K28A support

th0m4sek, 2022-12-25:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Chip is getting more popular in cheap keyboards.

keyboard-magpie, 2022-12-25:
I can find no evidence this chip is even re-programmable.

tzarc, 2022-12-26:
It's an 8051-based MCU -- extremely unlikely to land in QMK.
If you can put in the legwork we'll consider it, but nobody on the QMK team is going to do it for you.

closed 2022-12-26 by tzarc

th0m4sek, 2023-01-06:
http://www.wch-ic.com/downloads/CH555DS1_PDF.html
For somebody with skill and time in future.

SpeedxThunder001, 2023-03-19:
I got a redragon k552 VS11K28A with a firmware updater idk how helpful it is but I am leaving it here just in case
https://drive.google.com/drive/folders/1DAzKW4imj75MJEAjrc9q3LndAMzdJKgb?usp=sharing

UltraBlackLinux, 2023-04-23:
Glorious is starting to sell their GMMK full keyboards with this chip. I sadly got one of those and I'm really disappointed, considering that the previous models worked perfectly with other chips. The on-board firmware does not even have any FN mappings and the software to configure it does not detect my keyboard in a VM (I'm on linux).

sigprof, 2023-04-23:
> Glorious is starting to sell their GMMK full keyboards with this chip.

This was known for quite some time (maybe not in the usual places where QMK is discussed though, because the upstream QMK does not support the Sonix chips which were used in previous revisions of those keyboards, therefore any support for them was (and still is) in the SonixQMK fork):

https://www.reddit.com/r/OpenRGB/comments/qr6jca/gmmk_rev4/

Unfortunately, that's the risk of buying a keyboard for which the manufacturer does not advertise QMK support in an official way — there is always a chance of a “new improved revision” where the manufacturer “optimized costs” by replacing the MCU with some different model that might no longer be supported in QMK.  Note that Glorious also sells several other GMMK keyboard models which are officially advertised as supporting QMK (and the code for them is actually present in the upstream version of QMK, and not in some manufacturer's fork).

UltraBlackLinux, 2023-04-24:
The issue  I'm facing is that the main control chip is not even listed as supported there...

dev-devinsky, 2023-07-15:
> I got a redragon k552 VS11K28A with a firmware updater idk how helpful it is but I am leaving it here just in case https://drive.google.com/drive/folders/1DAzKW4imj75MJEAjrc9q3LndAMzdJKgb?usp=sharing

Hey! Could you be so kind and refresh the link? I have a keyboard with this chip and want to give it a try : )

SpeedxThunder001, 2023-07-16:
@dev-devinsky you can check the firmware updater here: https://cdn.shopify.com/s/files/1/0309/3679/3226/files/REDRAGON_K552RGB-1_28A_NEW_SOFTWARE_20210526_1.zip?v=1655183740

ItxWeird, 2023-08-18:
Any updates on this by chance?

tzarc, 2023-08-19:
No. Locking as further discussion serves no purpose.