qmk/qmk_firmware #19433: [Bug] PMW3360 Trackball sensor unequal CPI on the master/slave halves

Onefabis, 2022-12-28:
### Describe the Bug

When  I set #define PMW33XX_CPI 6000 both senshors should be the same in the movement sensitivity. But for some reason the slave sensor moved slowly. I.e. slave cpi = master cpi / 2. Can't find the reason why this behaviour remains if I switch the master/slave halves. I'm using f401 with full duplex and SPLIT_HAND_PIN.

### Keyboard Used

fabis

### Link to product page (if applicable)

https://github.com/Onefabis/qmk_firmware/tree/Haptic_split/keyboards/fabis

### Operating System

Windows

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.                                                                                              
Ψ CLI version: 1.1.1                                                                                                                    
Ψ QMK home: C:/Users/Alex_Laptop/qmk_firmware                                                                                           
Ψ Detected Windows 10 (10.0.22000).                                                                                                     
Ψ Git branch: Haptic_split                                                                                                              
Ψ Repo version: 0.19.5                                                                                                                  
⚠ Git has unstashed/uncommitted changes.                                                                                                
Ψ - Latest Haptic_split: 2022-12-24 23:40:28 +0300 (036ad6bc8b) -- Update readme.md                                                     
Ψ - Latest upstream/master: 2022-12-28 05:29:19 +0900 (4c32e060e8) -- [Keyboard] Add XR63GL keyboard (#19380)                           
Ψ - Latest upstream/develop: 2022-12-27 20:29:52 +0000 (40fddcb1e5) -- Merge remote-tracking branch 'origin/master' into develop        
Ψ - Common ancestor with upstream/master: 2022-12-23 18:14:23 +0000 (b56c0fa674) -- Bump actions/stale from 6 to 7 (#19392)             
Ψ - Common ancestor with upstream/develop: 2022-12-23 18:14:23 +0000 (b56c0fa674) -- Bump actions/stale from 6 to 7 (#19392)            
Ψ All dependencies are installed.                                                                                                       
Ψ Found arm-none-eabi-gcc version 10.1.0                                                                                                
Ψ Found avr-gcc version 8.5.0                                                                                                           
Ψ Found avrdude version 7.0                                                                                                             
Ψ Found dfu-util version 0.11                                                                                                           
Ψ Found dfu-programmer version 0.7.2                                                                                                    
Ψ Submodules are up to date.                                                                                                            
Ψ Submodule status:                                                                                                                     
Ψ - lib/chibios: 2022-09-18 10:01:17 +0000 --  (0e9d558b5)                                                                              
Ψ - lib/chibios-contrib: 2022-10-03 18:09:41 +0200 --  (bb8356fb)                                                                       
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)                                                                            
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)                                                                                 
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)                                                                               
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)                                                                                 
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)                                                                                   
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

freznel10, 2023-01-03:
This might not apply exactly to your keyboard but this is how I implmented it in the Scylla Ballz Rev 2 . 

[LINK](https://github.com/freznel10/qmk_firmware/blob/refactor/keyboards/zerfstudios/douballz/douballz.c#L120-L145)

nopia-alexander, 2023-01-03:
Thanks @freznel10 Now I see that initial haptic buzzing is shorter on the slave side than on the master side. Defitinelly there is something wrong with the Uart setup :(

Onefabis, 2023-01-03:
Thanks @freznel10 Now I see that initial haptic buzzing is shorter on the slave side than on the master side. Defitinelly there is something wrong with the Uart setup :(

Onefabis, 2023-01-04:
@freznel10 thanks, with your code I found some kind of workaround:
```
void keyboard_post_init_user(void) {
    pointing_device_set_cpi(7200);
    pointing_device_set_cpi_on_side(false, 8120);
}
```
Only in that order pointing_device_set_cpi_on_side(true, 7200) didn't work and 8120 cpi on the right side. If I use 8100 cpi that sensor doesn't work.

christrotter, 2023-07-03:
I ran into similar issues, too.  Both PMW3360 & 3389 on RP2040.  Neither of those fixes did it for me.  

I put in a print check `uprintf("left cpi is: %d \n", pointing_device_get_cpi());` to test.
```c
Ψ Console Connected: christrotter handwired/arcboard_mk17 (FEED:0000:1)
christrotter:handwired/arcboard_mk17:1: left cpi is: 18034
christrotter:handwired/arcboard_mk17:1: matrix scan frequency: 793
christrotter:handwired/arcboard_mk17:1: left cpi is: 18034
christrotter:handwired/arcboard_mk17:1: left cpi is: 18034
christrotter:handwired/arcboard_mk17:1: left cpi is: 18034
christrotter:handwired/arcboard_mk17:1: left cpi is: 42034
christrotter:handwired/arcboard_mk17:1: left cpi is: 42034
christrotter:handwired/arcboard_mk17:1: left cpi is: 42034
```

USB plugged into left, the left trackball works briefly then ceases to work at all.
USB plugged into right, the left trackball doesn't work at all.
In any case, the right trackball is working fine.  I'm using as-per-docs pointing (split/combined) code.  Right side as mouse, left side as drag scroll (code straight out of docs).