qmk/qmk_firmware #19440: How to get in touch regarding a security concern

benharvie, 2022-12-28:
Hello 👋

I run a security community that finds and fixes vulnerabilities in OSS. A researcher (@iamnynan) has found a potential issue, which I would be eager to share with you.

Could you add a `SECURITY.md` file with an e-mail address for me to send further details to? GitHub [recommends](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository) a security policy to ensure issues are responsibly disclosed, and it would help direct researchers in the future.

Looking forward to hearing from you 👍

(cc @huntr-helper)

zvecr, 2022-12-28:
As a note to whoever implements, any `SECURITY.md` file changes should be made to <https://github.com/qmk/.github/blob/main/SECURITY.md> and not directly to the qmk_firmware repository.

tzarc, 2023-03-29:
Was a non-event with a PoC hijacked link in docs. Not a vulnerability.

closed 2023-03-29 by tzarc