qmk/qmk_firmware #19446: [Bug] NOOP keys kill combos

xJayMorex, 2022-12-30:
### Describe the Bug

The issue is that whenever a NOOP key is pressed down along with a valid combo, the combo doesn't fire.

Use case:

I am using an ortholinear split keyboard with left shift + right shift combo as caps lock toggle. The right shift is 2U with a POS keycap so the underlaying switches are KC_RSFT and XXXXXXX (using the right switch as a stabilizer because the pcb doesn't support 2U key in that position).

Expected behavior:

NOOP shouldn't interfere with combos because by definition the key should be ignored.

Notes:

I tried the combo with 1U keycaps (meaning the NOOP key is not pressed) and it works fine, so the problem is indeed the NOOP interfering with combos.

### Keyboard Used

keebio/viterbi/rev2

### Link to product page (if applicable)

https://keeb.io/products/viterbi-keyboard-pcbs-5x7-70-split-ortholinear

### Operating System

Microsoft Windows 11

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

xJayMorex, 2022-12-30:
**Proposed solution:**

```
static bool process_single_combo(combo_t *combo, uint16_t keycode, keyrecord_t *record, uint16_t combo_index) {
	if (keycode == KC_NO) {
		return true;
	}
	(...)
}
```

```
bool process_combo(uint16_t keycode, keyrecord_t *record) {
	// ignore noop keys during combos
	if (keycode == KC_NO) {
		return true;
	}
	(...)
}
```

johnwilmes, 2024-05-19:
This is a special case of #15911