qmk/qmk_firmware #19472: Keychron Q6 HID Compatibility with TESmart KVM

CV8R, 2023-01-01:
### Describe the Bug

I have a Keychron Q6 Ansi with Audio Encoder knob and trying to use it via a Tesmart KVM. The Media Keys of the Q6 did not work via the HID input on the KVM with the factory firmware or default from QMK github.  Building firmware and keymap and setting KEYBOARD_SHARED_EP = yes got the media keys working but whenever I enable the shared EP the Scroll Lock-Scroll Lock hotkey on the KVM stops functioning.  I have tried NKRO_ENABLE = no & MOUSEKEY_ENABLE = no & RAW_ENABLE = no but the media keys then stop functioning via the HID input but the hotkey works. Direct USB is okay via the KVM USB hub or direct into the PC - but obviously i lose the hotkeys to switch KVM inputs.

I have a cheap Dell & Mouse wireless USB combo with media keys and a Logitech G810 with media keys and both work with Media and Hotkeys via the same HID input on the KVM.

I added the Scroll Lock key KC_SCRL to the keymap.c as its not there in the default.

### Keyboard Used

keychron/q6/ansi_encoder

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-q6-qmk-custom-mechanical-keyboard

### Operating System

 Ubuntu 22.04.1 LTS (and Win 11)

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

At the request of @sigprof from Discord:

Working Dell Keyboard:

dmesg output

`
[  339.706665] usb 1-3: new full-speed USB device number 6 using xhci_hcd
[  340.039541] usb 1-3: New USB device found, idVendor=413c, idProduct=301d, bcdDevice= 0.09
[  340.039548] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  340.039551] usb 1-3: Product: Dell Universal Receiver
[  340.039552] usb 1-3: Manufacturer: Dell
[  340.064446] input: Dell Dell Universal Receiver as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-3/1-3:1.0/0003:413C:301D.0028/input/input66
[  340.123300] hid-generic 0003:413C:301D.0028: input,hidraw5: USB HID v1.11 Keyboard [Dell Dell Universal Receiver] on usb-0000:01:00.0-3/input0
[  340.131009] input: Dell Dell Universal Receiver Mouse as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-3/1-3:1.1/0003:413C:301D.0029/input/input67
[  340.131164] input: Dell Dell Universal Receiver Consumer Control as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-3/1-3:1.1/0003:413C:301D.0029/input/input68
[  340.191360] input: Dell Dell Universal Receiver System Control as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-3/1-3:1.1/0003:413C:301D.0029/input/input69
[  340.191439] hid-generic 0003:413C:301D.0029: input,hidraw6: USB HID v1.11 Mouse [Dell Dell Universal Receiver] on usb-0000:01:00.0-3/input1
[  340.196784] hid-generic 0003:413C:301D.002A: hiddev1,hidraw7: USB HID v1.11 Device [Dell Dell Universal Receiver] on usb-0000:01:00.0-3/input2
`

usbhid-dump

`
003:005:001:DESCRIPTOR         1672613760.352216
 05 01 09 02 A1 01 85 01 09 01 A1 00 05 09 19 01
 29 05 15 00 25 01 95 05 75 01 81 02 95 01 75 03
 81 01 05 01 09 30 09 31 09 38 15 81 25 7F 75 08
 95 03 81 06 C0 C0

003:005:000:DESCRIPTOR         1672613760.356094
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 95 01 75 08 81 01 95 05 75 01
 05 08 19 01 29 05 91 02 95 01 75 03 91 01 95 06
 75 08 15 00 25 91 05 07 19 00 29 91 81 00 05 0C
 09 00 15 80 25 7F 75 08 95 08 B1 02 C0

003:004:002:DESCRIPTOR         1672613760.358033
 06 00 FF 09 00 A1 01 85 01 09 00 15 00 26 FF 00
 75 08 95 1F 82 02 01 C0 06 01 FF 09 01 A1 01 85
 08 95 1F 75 08 15 00 26 FF 00 09 20 B1 03 C0

003:004:001:DESCRIPTOR         1672613760.361028
[... 275 lines cut ...]
 FF 7F 75 10 95 02 09 30 09 31 81 06 15 81 25 7F
 75 08 95 01 09 38 81 06 05 0C 0A 38 02 95 01 81
 06 C0 C0

001:005:002:DESCRIPTOR         1672614183.027685
 05 01 09 80 A1 01 85 01 19 81 29 83 15 00 25 01
 95 03 75 01 81 02 95 01 75 05 81 01 C0 05 0C 09
 01 A1 01 85 02 15 00 25 01 95 12 75 01 0A 83 01
 0A 8A 01 0A 92 01 0A 94 01 09 CD 09 B7 09 B6 09
 B5 09 E2 09 EA 09 E9 0A 21 02 0A 23 02 0A 24 02
 0A 25 02 0A 26 02 0A 27 02 0A 2A 02 81 02 95 01
 75 0E 81 01 C0

001:005:001:DESCRIPTOR         1672614183.030692
 06 00 FF 09 01 A1 01 09 02 15 00 26 FF 00 75 08
 95 40 81 02 09 03 15 00 26 FF 00 75 08 95 40 91
 02 C0

001:005:000:DESCRIPTOR         1672614183.033685
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 95 01 75 08 81 03 95 03 75 01
 05 08 19 01 29 03 91 02 95 01 75 05 91 03 95 06
 75 08 15 00 26 A4 00 05 07 19 00 29 A4 81 00 C0

`

sigprof, 2023-01-02:
Do you really need `KEYBOARD_SHARED_EP = yes`?  At least the HID report descriptors of your Dell keyboard look similar to what you should get from QMK with the default `KEYBOARD_SHARED_EP = no` and `NKRO_ENABLE = no` (actually that board even has an extra proprietary HID interface).  The Logitech board even seems to support NKRO, but it uses a significantly different approach for that (QMK sends a bitmap of pressed keys, but that Logitech board just has another array of keycodes, so it is not full NKRO, just 6+20).

So you should probably try this configuration:
```make
KEYBOARD_SHARED_EP = no
NKRO_ENABLE = no
VIA_ENABLE = no

EXTRAKEY_ENABLE = yes
MOUSEKEY_ENABLE = yes

```
(try also `MOUSEKEY_ENABLE = no` if it still does not work, although that Dell board has a mouse interface too).

If this still does not help, please also collect the output of `lsusb -v` from those devices; you may use these commands to limit the output:
```sh
lsusb -v -d 413c:301d   # for the Dell keyboard
lsusb -v -d 046d:c337   # for the Logitech keyboard
lsusb -v -d 3434:0161   # for Keychron Q6
```

CV8R, 2023-01-02:
You sir are a legend!  That fixed the issue, thank you. I tried every combination I could think of except for turning off VIA.  I can leave NRO_ENABLE = yes and it still works.  Actually, my rules.mk is as simple as:

`
ENCODER_MAP_ENABLE = yes

VIA_ENABLE = no
`

I assume Extra Keys, Mouse Keys, etc are set in the defaults for the keyboard keymap somewhere and not disabled unless set explicitly to = off.

Thanks again,
CV8R

Not being able to use VIA is a livable downside, but I am curious as to why VIA caused the issue?

sigprof, 2023-01-02:
So just enabling VIA breaks compatibility with that KVM?  That's somewhat strange, because your Dell receiver also has a vendor-specific HID interface, and it apparently works.

Maybe the problem is the ordering of those interfaces — the rawhid interface used by VIA always gets the interface number 1 when it is enabled, so the order of interfaces becomes: 0 — boot keyboard, 1 — rawhid (VIA), 2 — shared (extrakey, mouse, NKRO, …).  Unfortunately, that ordering cannot be changed without breaking compatibility with the existing VIA app (apparently some HID APIs don't give access to the usage pages for HID application collections, therefore the VIA app has the interface number 1 hardcoded instead of finding the interface dynamically by looking for the corresponding HID usage).

As for NKRO, note that just setting `NKRO_ENABLE = yes` does not enable sending NKRO reports (although it might have that effect with some QMK versions due to a bug in the ChibiOS-specific USB code, but recently that bug was fixed). Normally you need to use the `NK_ON` or `NK_TOGG` keycode to turn NKRO on (but that setting is persistent, so you should not need to redo it every time the keyboard gets turned on). You can use some key tester and check whether it actually shows more than 6 keys pressed at the same time, and then also check that your KVM recognizes its hotkeys in that mode (maybe it does not, in which case you won't be able to use NKRO with it).

CV8R, 2023-01-02:
"So just enabling VIA breaks compatibility with that KVM?" - Yes. Does appear so with my tests. Whenever I enable VIA the Media Keys Stop working. The scroll lock hotkey combo still works though.

WRT NKRO - I can confirm toggling via a keycode to enable NKRO stops the KVM working so it must not be enabled. I set it to off now in the makefile.

sigprof, 2023-01-02:
NKRO incompatibility is unfortunate, but apparently a common issue. Maybe the (6+20)KRO implementation used by your Logitech keyboard would be more compatible (that implementation probably always fills the 6KRO report, and uses the extra 20 bytes only when more than 6 keys are held down); but QMK in the NKRO mode stops using the 6KRO report completely and uses only the NKRO report, and that KVM probably looks only at the 6KRO report for its hotkeys.

BTW, about “toggling via a keycode to enable NKRO stops the KVM working” — do you mean that KVM stops even passing the keypresses to the computer, or that just the hotkey stops working?

CV8R, 2023-01-02:
If I toggle NKRO on via a key press assigned to NKRO Toggle, the KVM fails and attached computer fails to recognize any key press. It's like the KVM goes deaf.

sigprof, 2023-01-02:
So it seems that the KVM does not even try to pass the NKRO reports through… which is unfortunate, but probably nothing could be done about that (except maybe redesigning the NKRO support in QMK so that it still mostly uses the 6KRO report, but that might break other things — e.g., some systems may not like the apparent switching between multiple input devices that would happen in that case when pressing more than 6 normal keys at once).

At least now you have a working configuration (without VIA though), and the default QMK configuration at least gives a working basic keyboard with that KVM, so I suppose that the issue may be closed.

When developing XAP (an upcoming replacement for VIA and other stuff), maybe it could make sense to try making the XAP interface the last one, so that KVMs like that could at least see the shared interface with media keys first. But maybe other compatibility concerns would still prevent that solution.

closed 2023-01-02 by sigprof

CV8R, 2023-01-02:
I agree, I have a working system now. Given this keyboard is for writing emails, word docs and general office duties and I'm not a gamer pressing 6 keys at once isn't an issue for me.

Thanks for sorting it for me and hopefully this work has contributed to XAP and the project in some small way.

Now to figure out how to se the default RGB to a static color and make my scroll lock key highlight like the caps and num lock!

jonofe, 2023-12-30:
Hey guys,
thanks for sharing this information. It also solved my problem with TESmart KVM and Q3 Knob ISO. By changing VIA_ENABLE to 'no', I'm now able to use the Knob as well as the second layer of the function keys.

BUT, I realized, that the Reset via Control-j-z/y does not work anymore. I did the default installation of the build environment and compiled via the following command:

qmk compile -kb keychron/q3/iso_encoder -km no_via

no_via ist a copy the keychron-keymap with VIA_ENABLE set to 'no' in the rules.mk in the folder /qmk_firmware/keyboards/keychron/q3/iso_encoder/keymaps/no_via.

Did I something wrong? What do I have to change to get the reset option back.
When flashing the factory firmware from keychron, the functionality is back again.

Any help is very much  appreciated.

BR, André

adophoxia, 2023-12-30:
@jonofe 
> BUT, I realized, that the Reset via Control-j-z/y does not work anymore.

That's because the firmware files in `qmk/qmk_firmware` for your board doesn't have the code that doesn't have this functionality, unlike in Keychron's fork of QMK in their `playground` branch where it does exist.

One *easy* way of getting it back, just bind `EE_CLR` to any empty spot in one of your layers in your `keymap.c`.

jonofe, 2023-12-30:
@adophoxia 
Thanks for this quick and easy solution. 👍 
BTW: Did I loose anything else than this reset functionality?
Is it also possible to use Keychron fork as the basis for a new firmware compilation?

adophoxia, 2023-12-31:
@jonofe 
> BTW: Did I loose anything else than this reset functionality?

As far as I know besides the `Hold FN + J + Z for 4 seconds` eeprom reset, there is increased encoder functionality but that's it.
> Is it also possible to use Keychron fork as the basis for a new firmware compilation?

You can, but if you for some reason don't need those two things, you can always work off of the main QMK repo.

jonofe, 2024-01-01:
Is there any chance, that VIAL will also solve the problem of the MEDIA keys via the KVM switch.
I have setup the build environment and sucessfully compiled the Q3 VIAL default firmware.
With the default setup the problem with the media key remains. And I read, that I should not deactivate VIA in the rules.mk.
Is that only a recommendation or would VIAL not work at all if I would deactivate VIA as described above in this thread?

richard-browne, 2024-05-05:
Can someone help this noob build Q6 firmware with VIA_ENABLE = no? Which rules.mk file do I put this in?

keychron/q6/ansi_encoder/rules.mk
keychron/q6/ansi_encoder/keymaps/rules.mk
keychron/q6/ansi_encoder/keymaps/default/rules.mk
keychron/q6/ansi_encoder/keymaps/keychron/rules.mk
keychron/q6/ansi_encoder/keymaps/via/rules.mk

richard-browne, 2024-05-05:
> You don't need to edit anything. You can build and flash the default keymap.

Thanks.. so what keymap do keychron themselves build as the official firmware? Is it 'keychron' or 'via' ?? What do I lose by flashing the 'default' keymap?

I just want to disable VIA to get it working with my KVM. Disabling VIA is no loss to me since it doesn't work anyway (the via app/web site never detects the keyboard).

trollixx, 2026-09-06:
A very late follow-up, but hopefully relevant to anyone landing here with a TESmart KVM.

I had the same class of failure with a Kinesis Freestyle Edge RGB on a TESmart HKS0403A1U (now marketed as HKS403-P23). The keyboard powered up on the dedicated K/M port, but no keystrokes were forwarded. Worked fine on the USB pass-through ports. Like the Q6 here, it's a keyboard with more than one HID interface.

It was fixed by a TESmart firmware update. The firmware is `HKS0403A1U_KM_-SOFMU16-V015.bin`. TESmart does not publish firmware. You have to request it via their support portal.

Worth noting that their support initially told me this was "a hardware/protocol limitation and cannot be resolved through a KVM firmware update," and then sent firmware that resolved it.

After flashing, all three interfaces pass through natively. I can see boot mouse, boot keyboard, and the vendor HID carrying Consumer/System Control. Media keys still work, and switching no longer re-enumerates the keyboard.

If you're on a TESmart and worked around this by disabling VIA or trimming interfaces, it may be worth asking them for the current firmware.