qmk/qmk_firmware #19483: How to edit and compile a custom firmware.

Magniflex99, 2023-01-03:
### Issue Description

Hi all, I hope I'm not in the wrong section, if so, sorry for my inexperience. I finally decided to customize the Keychron Q2 firmware that I currently own, to add and unlock some interesting features, that is, unlock the remaining keycaps backlight modes, and add a backlight suspension mode.

#define RGB_MATRIX_KEYPRESSES
#define RGB_MATRIX_FRAMEBUFFER_EFFECTS
#define RGB_DISABLE_WHEN_USB_SUSPENDED

The problem for me is that I don't know where to put my hands, I thought I could do the operations using Visual Studio Code, but I think I'm off track.
Would anyone have the adequate amount of patience to explain to a noob exactly what to do?

waffle87, 2023-01-05:
VS Code is a good text editor, but you'll also need to get a QMK build environment setup. Follow the [newbs guide](https://docs.qmk.fm/#/newbs_getting_started) and I suggest joining the QMK discord server if you have further questions.

Magniflex99, 2023-01-05:
> VS Code is a good text editor, but you'll also need to get a QMK build environment setup. Follow the [newbs guide](https://docs.qmk.fm/#/newbs_getting_started) and I suggest joining the QMK discord server if you have further questions.

Thanks a lot for the suggestion, I didn't know there was a Discord server, it could be really useful!

closed 2023-04-07 by waffle87