qmk/qmk_firmware #19492: [Bug] Cannot compile Miryoku keymap for Kyria, firmware too large

mptrsen, 2023-01-03:
### Describe the Bug

Following the [Miryoku compilation instructions](https://github.com/manna-harbour/miryoku_qmk/tree/miryoku/users/manna-harbour_miryoku), the compilation fails after finding that the firmware is too large:

    $ qmk compile -c -kb splitkb/kyria -km manna-harbour_miryoku
    Ψ Compiling keymap with gmake --jobs=1 splitkb/kyria/rev2:manna-harbour_miryoku
    
    Making splitkb/kyria/rev2 with keymap manna-harbour_miryoku and target clean
    
    Making splitkb/kyria/rev2 with keymap manna-harbour_miryoku

    avr-gcc (GCC) 5.4.0
    Copyright (C) 2015 Free Software Foundation, Inc.
    This is free software; see the source for copying conditions.  There is NO
    warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

    Generating: .build/obj_splitkb_kyria_rev2/src/info_config.h                                         [OK]
    Generating: .build/obj_splitkb_kyria_rev2/src/default_keyboard.c                                    [OK]
    Generating: .build/obj_splitkb_kyria_rev2/src/default_keyboard.h                                    [OK]
    Compiling: keyboards/splitkb/kyria/kyria.c                                                          [OK]
    [... more successful compiling messages ...]
    Linking: .build/splitkb_kyria_rev2_manna-harbour_miryoku.elf                                        [OK]
    Creating load file for flashing: .build/splitkb_kyria_rev2_manna-harbour_miryoku.hex                [OK]
    Copying splitkb_kyria_rev2_manna-harbour_miryoku.hex to qmk_firmware folder                         [OK]
    Checking file size of splitkb_kyria_rev2_manna-harbour_miryoku.hex                                 
     * The firmware is too large! 28764/28672 (92 bytes over)
     [ERRORS]

    gmake[1]: *** [builddefs/common_rules.mk:391: check-size] Error 1

Expected: Compilation produces a firmware that will fit.

This happens on both the master and miryoku branches.

### Keyboard Used

splitkb/kyria

### Link to product page (if applicable)

https://splitkb.com/products/kyria-pcb-kit

### Operating System

Ubuntu GNU/Linux

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /home/malte/qmk_firmware
Ψ Detected Linux.
Ψ Git branch: miryoku
Ψ - Latest miryoku: 2022-11-29 12:05:46 +1100 (d92f5d1772) -- [miryoku] Update changed QMK keycodes
Ψ - Latest upstream/master: 2023-01-02 17:51:32 -0500 (5962ebaba1) -- Fix github URL for the atreyu maintainer. (#19476)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2022-11-28 00:46:40 -0500 (d27b852fcc) -- new keyboard rb1 (#18164)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.6.1
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-09-18 10:01:17 +0000 --  (0e9d558b5)
Ψ - lib/chibios-contrib: 2022-10-03 18:09:41 +0200 --  (bb8356fb)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

tzarc, 2023-01-03:
You'll need to find stuff to disable, it seems.

QMK ensures the `default` keymap compiles correctly -- there are nearly 16000 build targets and we need to pick our battles.

keyboard-magpie, 2023-01-03:
Miryoku has its own dedicated help area; this would be better addressed there: https://github.com/manna-harbour/miryoku/discussions

closed 2023-01-03 by keyboard-magpie

mptrsen, 2023-01-03:
Ok. Thanks for your quick responses! I'll move the issue to the Miryoku discussion area.