qmk/qmk_firmware #19503: [Bug] Ergodox Infinity slave-half freezes soon after init

Hylian, 2023-01-04:
### Describe the Bug

I was bringing my Ergodox Infinity QMK firmware up to tip of master, which I had previously last updated in 2019. I found that after rebasing, the slave-half would work for a few seconds, and then freeze up, while the master-half would still operate correctly. I suspect it was freezing/crashing entirely, as the LCD backlight would stay lit, instead of turning off after a timeout as usual. When I swapped the master and slave halves, the new slave would still show the same crashing behavior.

I originally thought this was an issue with signal integrity of the serial connection, and messed around with GPIO drive strength, turning off LCD, various `SPLIT` settings, cables, etc., but to no avail. I finally decided to do a `git bisect` and found the offending commit where this issue was first introduced:

```
qmk_firmware (c4ff819) (BISECTING) via ⍱ 
❯ git bisect good
6f13a76530165bb1ad723ab0270c9eb908ca3a8c is the first bad commit
commit 6f13a76530165bb1ad723ab0270c9eb908ca3a8c
Author: Stefan Kerkmann <karlk90@pm.me>
Date:   Sun Oct 2 15:35:33 2022 +0200

    [Core] ChibiOS: Fix USB bus disconnect handling (#18566)

 tmk_core/protocol/chibios/usb_main.c | 3 ++-
 tmk_core/protocol/chibios/usb_util.c | 1 +
 2 files changed, 3 insertions(+), 1 deletion(-)
```

I haven't looked into it further, but I've been using the commit previous on my Infinity successfully.

### Keyboard Used

input_club/ergodox_infinity

### Operating System

Linux

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

zvecr, 2023-01-05:
Duplicate of #19420.

Hylian, 2023-01-05:
Commented on the original ticket; closing this one as dupe.

closed 2023-01-05 by Hylian