qmk/qmk_firmware #19577: `get_mods()` not returning Mod-Tap modifier status

seedrickMCWW, 2023-01-12:
### Issue Description

In my keymap, I use mod-tap on my home row--things like `LSFT_T(KC_T)`. It works properly but I'm also trying to display the modifier status on my OLEDs. I can write to the OLEDs just fine but I can't get anything when I poll the modifier status using `get_mods` against `MOD_MASK_CTRL`, `MOD_MASK_SHIFT`, `MOD_MASK_ALT`, and `MOD_MASK_GUI`. Is there another function or other masks I should be using instead?

sigprof, 2023-01-12:
`get_mods()` should work with mod-taps just fine (but only after the key action is actually determined to be a hold, which happens after the tapping term had expired, or maybe slightly earlier in some non-default modes if other keys are also pressed).

However, if you are using a split keyboard and trying to display the modifier state on the slave side, you need to add this to your `config.h` file:
```c
#define SPLIT_MODS_ENABLE
```
See [Split Keyboard — Data Sync Options](https://docs.qmk.fm/#/feature_split_keyboard?id=data-sync-options) for the documentation of that option and other sync options that you may want to use.

seedrickMCWW, 2023-01-12:
...Naturally, it was commented out of my code, because why enable things you need?

Thanks, that's everything solved.

closed 2023-01-12 by seedrickMCWW