qmk/qmk_firmware #19587: [Feature Request] Pointing Device Acceleration

gyvess, 2023-01-13:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

### Description

I'm calling this "Pointing Device Acceleration" solely to differentiate it from the already present Mouse Key Acceleration.

Pointing Device Acceleration, or Mouse Acceleration, is when the physical speed at which you move your pointing device influences the distance your cursor moves. For example, acceleration may allow users to slowly move their mouse 20cm over 5 seconds to move their cursor 200px and also quickly move their mouse 20cm over 1 second to move their cursor 1000px.

With an increasing number of pointing devices making use of QMK this would be a nice feature to have baked into the firmware. It would increase portability as users could simply set their sensitivity in the operating system/program without worrying if alternative methods of acceleration are available.

##### Two existing implementations for your consideration:

- https://github.com/NicolasFirmo/leetmouse. A fork of Linux USB driver that allows acceleration. I'm linking specifically to this one because it includes additional variables for curve customization.
- https://github.com/a1xd/rawaccel. Windows only but the most popular program for installing acceleration today.

Alabastard-64, 2023-01-13:
I already have been working on an implementation of different acceleration
schemes from rawaccel with the original goal being to add it as a pointing
device feature, However I have noticed that there is different acceleration
and inertial cursor code all over the place (mouse keys, gestures etc.).
These could use some unification/consistency but that requires a deeper
convo with the QMK team. So I shelved it until I finish adding the pointing
modes and hires scrolling features (in near final and draft PR state
respectively).
Also anyone who also may have started work on this let me know so we can
coordinate etc.

Obviously the trick is making sure to stick to fixed point implementations
of complex functions and trying to keep it as performant on Atmel U2/U4
chips as possible.  Arm chips especially ones with integer division
hardware such as the RP2040 will be more than fine as long as there is no
floating point math as none of these chips support it natively (as far as
I'm aware unless a cortex-M4 chip is used somewhere) so go to expensive and
slow emulation.

closed 2023-05-14 by github-actions[bot]

iamim, 2023-12-13:
I believe it to still be relevant. It could be a good alternative to the "sniping mode".