qmk/qmk_firmware #19600: [Bug] Adafruit RP2040 Macropad key stuck after Windows resume from sleep

ForsakenRei, 2023-01-15:
### Describe the Bug

I have an [Adafruit RP2040 Macropad](https://www.adafruit.com/product/5128) and tried different build recently while the last issue of media key was fixed, there is still a problem when Windows resume from sleep. 

After PC resume from sleep, the first key I pressed on the macropad will get stuck, for example I have `KC_F13` on it and I pressed it then it will be held until I reset the macropad or reconnect it. It basically become unresponsible after the first key press.

I'm on the lastest dev branch

### Keyboard Used

adafruit/macropad

### Link to product page (if applicable)

https://www.adafruit.com/product/5128

### Operating System

Windows 10 22H2

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.                                                      
Ψ CLI version: 1.1.0                                                                            
Ψ QMK home: C:/Users/Shigure/qmk_firmware                                                       
Ψ Detected Windows 10 (10.0.19045).                                                             
Ψ Git branch: develop                                                                           
Ψ Repo version: 0.19.9                                                                          
⚠ Git has unstashed/uncommitted changes.                                                        
Ψ - Latest develop: 2023-01-14 04:24:54 -0600 (45851a10f6) -- Add RGB565 and RGB888 color suppor
t to Quantum Painter (#19382)                                                                   
Ψ - Latest upstream/master: 2023-01-14 02:25:07 +0000 (3f2cbc9e68) -- Pr/reorder docker podman c
heck (#19561)                                                                                   
Ψ - Latest upstream/develop: 2023-01-15 15:04:32 +0000 (0730740add) -- Add mmoskal/uf2-stm32f103
 bootloader support (#19594)                                                                    
Ψ - Common ancestor with upstream/master: 2023-01-14 02:25:07 +0000 (3f2cbc9e68) -- Pr/reorder d
ocker podman check (#19561)                                                                     
Ψ - Common ancestor with upstream/develop: 2023-01-14 04:24:54 -0600 (45851a10f6) -- Add RGB565 
and RGB888 color support to Quantum Painter (#19382)                                            
Ψ All dependencies are installed.                                                               
Ψ Found arm-none-eabi-gcc version 10.1.0                                                        
Ψ Found avr-gcc version 8.5.0                                                                   
Ψ Found avrdude version 6.4                                                                     
Ψ Found dfu-programmer version 0.7.2                                                            
Ψ Found dfu-util version 0.11                                                                   
Ψ Submodules are up to date.                                                                    
Ψ Submodule status:                                                                             
Ψ - lib/chibios: 2023-01-03 19:29:26 +0000 --  (0062927e3)                                      
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)                               
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)                                    
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)                                         
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)                                           
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)                                         
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)                                       
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)                                          
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

N/A

### Additional Context

The unstashed changes come from the doc translation I haven't finish yet.

Maybe related to https://github.com/qmk/qmk_firmware/issues/18591? It's a different board so I opened a separate issue.

scchow, 2023-01-19:
I have also been seeing this issue with another board I've been working on. 

One thing I noticed was that is that it stopped after I disabled my RGB. 

Can you try setting `RGB_MATRIX_ENABLE = no` in your [`rules.mk`](https://github.com/qmk/qmk_firmware/blob/master/keyboards/adafruit/macropad/rules.mk) and see if the stuck key still occurs? If so, then we are probably encountering the same problem, which can help narrow the issue down.

ForsakenRei, 2023-01-20:
> I have also been seeing this issue with another board I've been working on. I've posted a writeup of my observations in this [comment](https://github.com/qmk/qmk_firmware/issues/16969#issuecomment-1366934566).
> 
> One thing I noticed was that is that disabling my RGB fixes the issue. (Albeit not a great solution because I like RGB :))
> 
> Can you try turning off the RGB by setting `RGB_MATRIX_ENABLE = no` in your [`rules.mk`](https://github.com/qmk/qmk_firmware/blob/master/keyboards/adafruit/macropad/rules.mk) and see if the stuck key still occurs? If so, then we are probably encountering the same problem, which can help narrow the issue down.
> 
> I've been seeing this issue on a split keyboard (Sofle with Elite-Pi controller) using `RGB_LIGHT` instead of `RGB_MATRIX` so if we are running into the same problem. then the problem may be more widespread than my relatively niche setup.

That's quite interesting, I will give it a try and see if RGB is the trick.

ForsakenRei, 2023-01-20:
@scchow yes once I disable the RGB everything works just fine upon resume from sleep. Quite interesting though, I will proabably resync the repo sometime later today and see if anything changed.

scchow, 2023-01-20:
Good to know! I'll be doing some more debugging myself this weekend.

I started noticing this problem after I made some significant changes to my
keymap, so hopefully some regression tests can show which commit/feature
broke it.

Can you link your keymap/config files? It would be nice to cross reference
what features we both have enabled to help narrow things down.

ForsakenRei, 2023-01-20:
> Good to know! I'll be doing some more debugging myself this weekend. I started noticing this problem after I made some significant changes to my keymap, so hopefully some regression tests can show [...]

https://github.com/ForsakenRei/RP2040-MacroPad here's mine, I didn't update the `rules.mk` yet.

KarlK90, 2023-01-21:
@ForsakenRei @scchow could you please check if my fixes to the ws2812 vendor driver in https://github.com/qmk/qmk_firmware/pull/19652 fix your deadlocks.

scchow, 2023-01-21:
> could you please check if my fixes to the ws2812 vendor driver in #19652 fix your deadlocks.

Good catch on the timer overflow into deadlock! I've merged your fix into my branch and am currently testing it now. (Unfortunately have to leave my board idle for a while to see if it works.)

In the meantime, I've been working on a separate issue where keypresses from my rp2040 board doesn't wake Windows after it goes to S3 sleep (similar to #19008). 

I thought it might have had the same root cause as this issue, but now it seems like it may be something entirely different.

Thanks for your help!

npross, 2023-01-21:
I've recently completed my sofle build with kb2040 controllers and I have seen issues that sound very similar.  My list of problems have been:  Can't always wake my laptop from sleep, and other unresponsive keyboard issues from boot, sleep etc... I look forward to your findings.  I don't feel I can be much help actually diagnosing these issues but I certainly can apply patches and do some testing.  I have already applied @KarlK90 fix to see if some of my issues go away.   Thanks for your efforts!

scchow, 2023-01-22:
@KarlK90 Can confirm that i am no longer seeing the stuck key issue after applying your patch. Thanks!

I'll make a separate issue for the wake from S3 sleep issue.

ForsakenRei, 2023-01-23:
> @ForsakenRei @scchow could you please check if my fixes to the ws2812 vendor driver in #19652 fix your deadlocks.

Just got back home and it seems it all works for others, I will give it a try as well. Thanks for your effort!

KarlK90, 2023-01-23:
Thanks everyone for testing #19652 it has now been merged into develop. @ForsakenRei please close this issue if this resolves your issue.

ForsakenRei, 2023-01-24:
> Thanks everyone for testing #19652 it has now been merged into develop. @ForsakenRei please close this issue if this resolves your issue.

I think the key stuck issue was solved. I didn't have the S3 sleep issue but do encountered some weird RGB issue, I will do some test before open a new issue.

closed 2023-01-24 by ForsakenRei