qmk/qmk_firmware #19604: Can't compile with joystick config

seedrick, 2023-01-16:
### Issue Description

I'm trying to follow the instructions for a joystick mapping in the docs. Everything seems ok until I try to map the axes (I'm not using any actual analog):

`note: previous declaration of ‘joystick_axes’ was here
 extern joystick_config_t joystick_axes[JOYSTICK_AXIS_COUNT];`

Here's what I have in the struct:
`// Joystick Config
joystick_config_t joystick_axes[JOYSTICK_AXES_COUNT] = {
    JOYSTICK_AXIS_VIRTUAL,
    JOYSTICK_AXIS_VIRTUAL,
    JOYSTICK_AXIS_VIRTUAL,
    JOYSTICK_AXIS_VIRTUAL,
    JOYSTICK_AXIS_VIRTUAL,
    JOYSTICK_AXIS_VIRTUAL
};`

in my rules.mk:
`JOYSTICK_ENABLE = yes #gamepad with no actual analog input
JOYSTICK_DRIVER = digital`

In my config.h:
`//gamepad stuff
// 1000hz Polling Rate
#define USB_POLLING_INTERVAL_MS 1
#define QMK_KEYS_PER_SCAN 12

// Joystick Button Count
#define JOYSTICK_BUTTON_COUNT 32

// Joystick Axes Count
#define JOYSTICK_AXES_COUNT 6

// Joystick Axes Resolution
#define JOYSTICK_AXES_RESOLUTION 8`

Am I missing something or are the docs out of date?

sigprof, 2023-01-16:
Some configuration parameters for the joystick feature [had been renamed recently](https://docs.qmk.fm/#/ChangeLog/20221126?id=config-refactoring):

Old Config | New Config
-- | --
`JOYSTICK_AXES_COUNT` | `JOYSTICK_AXIS_COUNT`
`JOYSTICK_AXES_RESOLUTION` | `JOYSTICK_AXIS_RESOLUTION`

(if you don't see that on `docs.qmk.fm`, force refresh the page — sometimes caching does strange things with that site).

However, some examples in docs still use the old names… which is a bug.

sigprof, 2023-01-16:
Made the PR to update the examples: #19605

Also note that `USB_POLLING_INTERVAL_MS` defaults to 1 now, and `QMK_KEYS_PER_SCAN` was removed completely, so apparently your config needs updating (maybe some obsolete guide was used).

seedrick, 2023-01-17:
Thanks! It was indeed outdated documentation.

closed 2023-01-17 by seedrick