qmk/qmk_firmware #19610: mouse randomly jerky on JJ40 rev 1.5

papourien, 2023-01-16:
### Issue Description

hello here, i'am new/nood in qmk world

i have a jj40, works with qmk configurator and via, but mouse function doesn'tr works
so i started making my own code, adding mousekey_enable = on in rules file, now the mouse mouve but verry jerky
so i tryed different mode (linear and other) and always the same problem, the mouse is verry jerky but when i use all direction, the mouse become realy soft until i stop the mouse and become jerky again

i have other issue with this board, on hub usb or some old usb port, i have a problem of "keylocked" on combo, like a simple shifted a become AAAAAAAAAAAAAAAAAA
same problem on gmk configurator with mod key "like shit on hold kc on tap", but the function works when i do the firmware my self so...

i saw similar problem with this board here, but no found the good solution for me

some one have amy idea ?
thanks.

gudvinr, 2023-01-25:
Looks similar to #11389

papourien, 2023-01-25:
i saw this thread...
i had sticky problem out of the box, and he's gone when i changed the usb port
i still had combo sticky problem, i solve it stoping using QMK configurator and compiling the firmware my self
what is my firmware ? i see QMK firmware 0.19.8 when i compile, this is it ?
now, the jj40 works good, i actived mouse control, except the mouse control is jerky, the mouse control need press in all direction quickly for moving smooth, and if i stop it, become jerky again

same on different computer, i try it on my android phone too, the mouse looks works well on my phone until freeze all the keyboard (less one minute)

closed 2023-05-26 by github-actions[bot]