qmk/qmk_firmware #19648: [Bug] Docker builds broken when both Podman and Docker are installed

leah-splitkb, 2023-01-20:
### Describe the Bug

#19561 recently got merged into `master`.

However, after pulling it, I now get the following output instead of a properly working build:
[image]
This requires additional keyboard interaction, and only selecting the `docker.io` image works - the other images result in either `manifest unknown` or `repo not found`.

To my surprise, it turned out that QMK suddenly started using Podman instead of Docker!

I am running Fedora 35. I believe Podman was automatically installed, but I manually installed Docker because - as #19561 mentions - Podman does not have feature-parity with Docker. This is probably not an unusual scenario.

#19561 reordered the scripts to give priority to Podman because `Podman does not support the Docker API 100%`. The _actual_ issue seemed to be that the scripts did not properly detect an _aliased_ Podman, which meant to it accidentally using Docker-intended flags with Podman. Checking for Podman first meant it was now using the proper flags, but it also means that it switched the Docker backend for the people who genuinely have both installed. This seems like a mistake to me - and if not it is a rather large change to suddenly dump into `master`.

Basically, it seems it fixed the case where `docker` is aliased to Podman, but broke the case where both Docker and Podman are installed. It probably needs to be reverted and redone with proper alias detection.

@-ing @zvecr because he processed the original PR.

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

sigprof, 2023-01-20:
Having both Docker and Podman installed is probably a less usual configuration than having just Podman installed with its `docker` alias, and any choice would be wrong for some users in that case.  But you can choose the runtime explicitly instead of relying on autodetection using one of these commands:
* `export RUNTIME=docker`
* `export RUNTIME=podman`

tzarc, 2023-01-25:
I'm unsure how the alias comes into play -- the script execution is done in an external invocation of the shell, which under normal circumstances shouldn't be executing the login scripts. The alias shouldn't exist at the time, right? Only way I'd imagine it to be there is if it's symlink'ed?