qmk/qmk_firmware #19671: [Bug] `AUTO_SHIFT_MODIFIERS` occasionally sends single key instead of with-modifier-variant

mplattner, 2023-01-23:
### Describe the Bug

Since I've enabled `AUTO_SHIFT_MODIFIERS`, I have the issue that ~5% of all uses/presses of e.g. `CTRL+c` (copy to clipboard) just send a `c` instead of `CTRL+c`.

I thought it might be that I unintentionally press `CTRL+c` for too long (longer than `AUTO_SHIFT_TIMEOUT`, which is 150ms in my case), and thus it sends `CTRL+SHIFT+c` (as expected, when you press it for too long), but this is not the case: it doesn't send `CTRL+SHIFT+c`, but actually only `c`.

Seems like a bug to me. [Here](https://github.com/mplattner/qmk_firmware/blob/master/keyboards/sculpter/keymaps/mplattner/config.h) is my keyboard code. Any ideas?

### Keyboard Used

sculpter

### Operating System

Windows 10

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

mplattner, 2023-02-04:
Thanks @lesshonor. It feels like the issue occurs less often now, but it's not fixed. Do you suggest do lower or increase the value of `TAP_CODE_DELAY` further?

mplattner, 2023-02-05:
Thanks again. Incorporating #19449 did not help. I did some more testing. It all boils down to whether `AUTO_SHIFT_MODIFIERS` is defined, i.e., enabled. If it is, the issue occurs; otherwise it does not.

The only two occurences of `AUTO_SHIFT_MODIFIERS` within qmk are https://github.com/qmk/qmk_firmware/blob/4020674163fc80914059c4c9c3be5c0ae00bd150/quantum/process_keycode/process_auto_shift.c#L157 and https://github.com/qmk/qmk_firmware/blob/4020674163fc80914059c4c9c3be5c0ae00bd150/quantum/process_keycode/process_auto_shift.c#L456
so I guess the issue is somewhere there?

mplattner, 2023-05-14:
Thanks @IsaacElenbaas. I tried to incorporate your suggestion in https://github.com/mplattner/qmk_firmware/commit/4eec608c552a98ec79b8b60a8a92c2b45dee51fb. However, it did not resolve the issue, it felt like it got worse. Any other ideas?

mplattner, 2023-08-15:
I've just updated to the latest version in the master branch (https://github.com/qmk/qmk_firmware/commit/dbd847d4d3835ec4ab1067b535231284c5cece2d), but the problem persists.
Any help/ideas are appreciated 😊

IsaacElenbaas, 2023-12-05:
The probably bad code shape I described above was changed for other reasons in the latest dev->master merge. Could you try once more? I would have guessed that the `TAP_CODE_DELAY` would have fixed this, but looking at the code now, `AUTO_SHIFT_MODIFIERS` should have no effect on how keys are handled.

mplattner, 2023-12-05:
Thanks for getting back! Would be great if this can be fixed.
Good question - my `CTRL` is actually a `OSM(MOD_LCTL)`, `SHIFT` is `OSM(MOD_LSFT)` - see https://github.com/mplattner/qmk_firmware/blob/4eec608c552a98ec79b8b60a8a92c2b45dee51fb/keyboards/sculpter/keymaps/mplattner/keymap.c#L47
I haven't tried how it behaves with the regular variant (just `KC_LCTL`).

Not sure if this is related or if it can be fixed together, but another problem that persists for me with `OSM()` is documented in https://github.com/qmk/qmk_firmware/issues/18488.

Thank you @IsaacElenbaas!