qmk/qmk_firmware #19680: [Bug] Compile errors in quantum/keymap_introspection.c

zone-id, 2023-01-25:
### Describe the Bug

When using qmk compile or qmk flash (with a keymap.bin from QMK Configurator) I get this errors:
```
Ψ Compiling keymap with make --jobs=1 keychron/q3/ansi_encoder:my-keychron-q3

QMK Firmware 0.19.10
Making keychron/q3/ansi_encoder with keymap my-keychron-q3

arm-none-eabi-gcc (Arch Repository) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: quantum/keymap_introspection.c                                                          quantum/keymap_introspection.c: In function 'encodermap_layer_count':
quantum/keymap_introspection.c:35:52: error: invalid application of 'sizeof' to incomplete type 'const uint16_t[][1][2]' {aka 'const short unsigned int[][1][2]'}
   35 | #    define NUM_ENCODERMAP_LAYERS ((uint8_t)(sizeof(encoder_map) / ((NUM_ENCODERS) * (2) * sizeof(uint16_t))))
      |                                                    ^
quantum/keymap_introspection.c:38:12: note: in expansion of macro 'NUM_ENCODERMAP_LAYERS'
   38 |     return NUM_ENCODERMAP_LAYERS;
      |            ^~~~~~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c: At top level:
quantum/keymap_introspection.c:35:52: error: invalid application of 'sizeof' to incomplete type 'const uint16_t[][1][2]' {aka 'const short unsigned int[][1][2]'}
   35 | #    define NUM_ENCODERMAP_LAYERS ((uint8_t)(sizeof(encoder_map) / ((NUM_ENCODERS) * (2) * sizeof(uint16_t))))
      |                                                    ^
quantum/keymap_introspection.c:41:37: note: in expansion of macro 'NUM_ENCODERMAP_LAYERS'
   41 | _Static_assert(NUM_KEYMAP_LAYERS == NUM_ENCODERMAP_LAYERS, "Number of encoder_map layers doesn't match the number of keymap layers");
      |                                     ^~~~~~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c:14:27: error: expression in static assertion is not an integer
   14 | #define NUM_KEYMAP_LAYERS ((uint8_t)(sizeof(keymaps) / ((MATRIX_ROWS) * (MATRIX_COLS) * sizeof(uint16_t))))
      |                           ^
quantum/keymap_introspection.c:41:16: note: in expansion of macro 'NUM_KEYMAP_LAYERS'
   41 | _Static_assert(NUM_KEYMAP_LAYERS == NUM_ENCODERMAP_LAYERS, "Number of encoder_map layers doesn't match the number of keymap layers");
      |                ^~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c: In function 'keycode_at_encodermap_location_raw':
quantum/keymap_introspection.c:35:52: error: invalid application of 'sizeof' to incomplete type 'const uint16_t[][1][2]' {aka 'const short unsigned int[][1][2]'}
   35 | #    define NUM_ENCODERMAP_LAYERS ((uint8_t)(sizeof(encoder_map) / ((NUM_ENCODERS) * (2) * sizeof(uint16_t))))
      |                                                    ^
quantum/keymap_introspection.c:44:21: note: in expansion of macro 'NUM_ENCODERMAP_LAYERS'
   44 |     if (layer_num < NUM_ENCODERMAP_LAYERS && encoder_idx < NUM_ENCODERS) {
      |                     ^~~~~~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c: In function 'encodermap_layer_count':
quantum/keymap_introspection.c:39:1: error: control reaches end of non-void function [-Werror=return-type]
   39 | }
      | ^
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [builddefs/common_rules.mk:360: .build/obj_keychron_q3_ansi_encoder_my-keychron-q3/quantum/keymap_introspection.o] Error 1
Make finished with errors
make: *** [Makefile:414: keychron/q3/ansi_encoder:my-keychron-q3] Error 1
```
Initial research suggests, that I am using a compiler unfit for this C code? I had installed avr-gcc v8.5, since the qmk doctor warned me about a later version, but that didn't fix it, so I reverted the change.

### Keyboard Used

keychron/q3/ansi_encoder

### Operating System

Garuda (Arch)

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /home/phil/repos/qmk_firmware
Ψ Detected Linux.
Ψ Git branch: master
Ψ Repo version: 0.19.10
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-09-18 10:01:17 +0000 --  (0e9d558b5)
Ψ - lib/chibios-contrib: 2022-10-03 18:09:41 +0200 --  (bb8356fb)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

zvecr, 2023-01-25:
You will need to post your `keymap.c` as this would be an issue within that file.

zone-id, 2023-01-25:
```
#include QMK_KEYBOARD_H

/* THIS FILE WAS GENERATED!
 *
 * This file was generated by qmk json2c. You may or may not want to
 * edit it directly.
 */

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
	[0] = LAYOUT_tkl_f13_ansi(KC_ESC, KC_BRID, KC_BRIU, KC_NO, LGUI(KC_A), RGB_VAD, RGB_VAI, KC_MPRV, KC_MPLY, KC_MNXT, KC_MUTE, KC_VOLD, KC_VOLU, KC_MUTE, KC_PSCR, KC_NO, RGB_TOG, KC_GRV, KC_1, KC_2, KC_3, KC_4, KC_5, KC_6, KC_7, KC_8, KC_9, KC_0, KC_MINS, KC_EQL, KC_BSPC, KC_INS, KC_HOME, KC_PGUP, KC_TAB, KC_Q, KC_W, KC_E, KC_R, KC_T, KC_Z, KC_U, KC_I, KC_O, KC_P, KC_LBRC, KC_RBRC, KC_BSLS, KC_DEL, KC_END, KC_PGDN, KC_CAPS, KC_A, KC_S, KC_D, KC_F, KC_G, KC_H, KC_J, KC_K, KC_L, KC_SCLN, KC_QUOT, KC_ENT, KC_LSFT, KC_Y, KC_X, KC_C, KC_V, KC_B, KC_N, KC_M, KC_COMM, KC_DOT, KC_SLSH, KC_RSFT, KC_UP, LCTL_T(OSL(4)), KC_LGUI, KC_LOPT, KC_SPC, KC_ROPT, KC_RGUI, MO(1), KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT),
	[1] = LAYOUT_tkl_f13_ansi(KC_TRNS, KC_F1, KC_F2, KC_F3, KC_F4, KC_F5, KC_F6, KC_F7, KC_F8, KC_F9, KC_F10, KC_F11, KC_F12, RGB_TOG, KC_TRNS, KC_TRNS, RGB_TOG, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, RGB_HUI, RGB_SAI, RGB_SPI, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, RGB_HUD, RGB_SAD, RGB_SPD, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, NK_TOGG, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, RGB_VAI, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, RGB_RMOD, RGB_VAD, RGB_MOD),
	[2] = LAYOUT_tkl_f13_ansi(KC_ESC, KC_F1, KC_F2, KC_F3, KC_F4, KC_F5, KC_F6, KC_F7, KC_F8, KC_F9, KC_F10, KC_F11, KC_F12, KC_MUTE, KC_PSCR, KC_NO, RGB_MOD, KC_GRV, KC_1, KC_2, KC_3, KC_4, KC_5, KC_6, KC_7, KC_8, KC_9, KC_0, KC_MINS, KC_EQL, KC_BSPC, KC_INS, KC_HOME, KC_PGUP, KC_TAB, KC_Q, KC_W, KC_E, KC_R, KC_T, KC_Y, KC_U, KC_I, KC_O, KC_P, KC_LBRC, KC_RBRC, KC_BSLS, KC_DEL, KC_END, KC_PGDN, KC_CAPS, KC_A, KC_S, KC_D, KC_F, KC_G, KC_H, KC_J, KC_K, KC_L, KC_SCLN, KC_QUOT, KC_ENT, KC_LSFT, KC_Z, KC_X, KC_C, KC_V, KC_B, KC_N, KC_M, KC_COMM, KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_LCTL, KC_LGUI, KC_LALT, KC_SPC, KC_RALT, KC_RGUI, MO(3), KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT),
	[3] = LAYOUT_tkl_f13_ansi(KC_TRNS, KC_BRID, KC_BRIU, KC_TRNS, LGUI(KC_A), RGB_VAD, RGB_VAI, KC_MPRV, KC_MPLY, KC_MNXT, KC_MUTE, KC_VOLD, KC_VOLU, RGB_TOG, KC_TRNS, KC_TRNS, RGB_TOG, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, RGB_TOG, RGB_MOD, RGB_VAI, RGB_HUI, RGB_SAI, RGB_SPI, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, RGB_RMOD, RGB_VAD, RGB_HUD, RGB_SAD, RGB_SPD, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, NK_TOGG, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS)
};
```

zvecr, 2023-01-25:
You have `ENCODER_MAP` enabled, but are not specifiying any `encoder_map` config?

<https://docs.qmk.fm/#/feature_encoders?id=encoder-map>

zone-id, 2023-01-26:
Well it seems there were a lot of things I didn't consider. I thought i could simply use the QMK configurator and use the JSON file to flash it to my keyboard, but that wasn't the case, due to the lack of encoder_map support.

closed 2023-01-26 by zone-id