qmk/qmk_firmware #19684: [Bug] Error compiling custom KB (Master 2023-01-25)

mnesarco, 2023-01-25:
### Describe the Bug

I just cloned master and installed qmk on a new laptop. Then when I try to compile my custom keyboard firmware, it fails with this error:

```

Compiling: tmk_core/protocol/usb_descriptor.c                                                      In file included from tmk_core/protocol/usb_descriptor.c:42:
tmk_core/protocol/usb_descriptor_common.h:21:29: error: 'LMarz' undeclared here (not in a function)
   21 | #define USBSTR(s) USBCONCAT(L, s)
      |                             ^
tmk_core/protocol/usb_descriptor_common.h:20:25: note: in definition of macro 'USBCONCAT'
   20 | #define USBCONCAT(a, b) a##b
      |                         ^
tmk_core/protocol/usb_descriptor.c:1065:42: note: in expansion of macro 'USBSTR'
 1065 |         .Size                   = sizeof(USBSTR(MANUFACTURER)),
      |                                          ^~~~~~
tmk_core/protocol/usb_descriptor_common.h:21:29: error: 'LSplitKeyboardDDK' undeclared here (not in a function)
   21 | #define USBSTR(s) USBCONCAT(L, s)
      |                             ^
tmk_core/protocol/usb_descriptor_common.h:20:25: note: in definition of macro 'USBCONCAT'
   20 | #define USBCONCAT(a, b) a##b
      |                         ^
tmk_core/protocol/usb_descriptor.c:1073:42: note: in expansion of macro 'USBSTR'
 1073 |         .Size                   = sizeof(USBSTR(PRODUCT)),
      |                                          ^~~~~~
 [ERRORS]
 | 
 | 
 | 
gmake[1]: *** [builddefs/common_rules.mk:360: .build/obj_handwired_marz44_default/usb_descriptor.o] Error 1
Make finished with errors

```

### Keyboard Used

custom_xxx

### Operating System

Linux

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /home/userx/devel/projects/Marz/kbd/qmk_firmware
Ψ Detected Linux.
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp /home/userx/devel/projects/Marz/kbd/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp /home/userx/devel/projects/Marz/kbd/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32-dfu' boards. Run 'sudo cp /home/userx/devel/projects/Marz/kbd/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp /home/userx/devel/projects/Marz/kbd/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp /home/userx/devel/projects/Marz/kbd/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp /home/userx/devel/projects/Marz/kbd/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /home/userx/devel/projects/Marz/kbd/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp /home/userx/devel/projects/Marz/kbd/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'md-boot' boards. Run 'sudo cp /home/userx/devel/projects/Marz/kbd/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Detected ModemManager without the necessary udev rules. Please either disable it or set the appropriate udev rules if you are using a Pro Micro.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp /home/userx/devel/projects/Marz/kbd/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/userx/devel/projects/Marz/kbd/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Git branch: master
Ψ Repo version: 0.19.10
Ψ - Latest master: 2023-01-25 00:19:32 +0000 (fa132baa43) -- Decrease LED animation frequency to improve performance (#19677)
Ψ - Latest upstream/master: 2023-01-25 00:19:32 +0000 (fa132baa43) -- Decrease LED animation frequency to improve performance (#19677)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-01-25 00:19:32 +0000 (fa132baa43) -- Decrease LED animation frequency to improve performance (#19677)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.6.1
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-09-18 10:01:17 +0000 --  (0e9d558b5)
Ψ - lib/chibios-contrib: 2022-10-03 18:09:41 +0200 --  (bb8356fb)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

It works in my other PC with an older qmk version:
```

Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: /home/userx/devel2/projects/active/Marz/projects/kbd/qmk_firmware_master
Ψ Detected Linux.
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp
/home/userx/devel2/projects/active/Marz/projects/kbd/qmk_firmware_master/util/udev/50-qmk.rules
/etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp
/home/userx/devel2/projects/active/Marz/projects/kbd/qmk_firmware_master/util/udev/50-qmk.rules
/etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp
/home/userx/devel2/projects/active/Marz/projects/kbd/qmk_firmware_master/util/udev/50-qmk.rules
/etc/udev/rules.d/'.
Ψ Git branch: master
Ψ Repo version: 0.18.5
Ψ - Latest master: 2022-09-05 21:26:11 -0700 (712aa7b1cd) -- Mars 6.5
Layout Additions (#18282)
Ψ - Latest upstream/master: 2023-01-25 00:19:32 +0000 (fa132baa43) --
Decrease LED animation frequency to improve performance (#19677)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2022-09-05 21:26:11 -0700
(712aa7b1cd) -- Mars 6.5 Layout Additions (#18282)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 6.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.6.1
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-06-11 09:09:45 +0000 --  (f836d24b0)
Ψ - lib/chibios-contrib: 2022-07-27 10:46:15 +0200 --  (d03aa9cc)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-05-17 19:19:01 +0200 --  (07edde8)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

mnesarco, 2023-01-25:
Maybe related: https://github.com/qmk/qmk_firmware/pull/19147#issuecomment-1326970159

keyboard-magpie, 2023-01-25:
Looks like you are declaring your usb identifiers wrongly- they need to be in info.json, inside quotes- see extant keyboards in the repository for examples.

mnesarco, 2023-01-25:
Ok, in previous version quotes were not required. But it fixed the problem.

```
// in config.h
#define MANUFACTURER    "Marz"
#define PRODUCT         "Split Keyboard DDK"
```

mnesarco, 2023-01-25:
@keyboard-magpie Thank you.

closed 2023-01-25 by mnesarco

mnesarco, 2023-01-25:
Ok, Thank you.
It is hard to follow all the changes, there is not backwards compatibility here. Every time I try to upgrade some of my old (6 months old) keyboards something that previously worked is broken :D.

Thank you for the info, I will move that configuration to info.json

keyboard-magpie, 2023-01-25:
Consider submitting a Pull Request for your keyboard as a way of keeping up with QMK's regular breaking changes.

mnesarco, 2023-01-25:
My Keyboards are just for personal use, nothing important to pollute qmk repo...
[image]

BTW, thank you for the amazing firmware.