qmk/qmk_firmware #19719: [Bug]

alxchk, 2023-01-30:
### Describe the Bug

RGB for the second half (i.e. non-master) of keyboad does not always turned off. I observed this RGB timeout, however this also happens with RGB_TOG. It's not always reproduced (I can't find pattern). Even presing RGB_TOG many times sometimes works and sometimes not. Not clear how to troubleshoot.

### Keyboard Used

keebio/iris/rev7

### Operating System

Linux

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

waffle87, 2023-01-30:
Its inconsistency and the fact it does occasionally work makes this sound like a hardware issue. Does it happen on both halves if they're the secondary half?

alxchk, 2023-01-30:
Yes. I primary is left, this may happen with right. If primary is right, then -- with left.
This doesn't feel like hardware issue, more like a bug with matrix driver, however I'm not an expert.

myst729, 2023-01-31:
Maybe double check the cable between two halves?

alxchk, 2023-02-01:
After some time with RGB_MATRIX_TIMEOUT I can add that this happens at the master as well as at with the slave.
Maybe this have some relation to what keys has been pressed last.

VictorSCamargo, 2025-03-11:
I have the exact same issue.

I have build many identical keyboards, using different controllers and cables, and it still happens. I even changed the communication speed because I was concerned that it could be some sort of interference that could be making the oposite side receive a bad communication.

But after all that, it still works.

But now I have an important question for you: do you use Vial? 
Because I am using it and did not test it yet with default QMK.

alxchk, 2025-03-11:
No, I don't use vial at all. Pure QMK.

VictorSCamargo, 2025-03-11:
Lets check together a couple more things if you are intested in investigating this:
- I build my custom version of Sofle RGB that uses USB C to connect halves (as Iris also uses)
- I am using a 90 degrees USB cable
- Even with USB I am using the default half duplex communication
- I have set a lower bit rate for communication
- I use RGB matrix
- I have asked this question before in the QMK discord and the only suggestions that I got was related to lowering bit rate

Your keyboard was bought ready to go, right? You didnt build it I guess. I will try to contact the creator

My source code can be found [here](https://github.com/TergoTeclados/vial-qmk-firmware).

alxchk, 2025-03-11:
I have Iris 7, with pre-soldered ATmega32u4, and I think I really bruteforced most possible options which were available. I gave up on that more than 2 years ago, and don't remember exactly what I had investigated. What I remember that I came to decision that it's a bit too much for me to debug it completely, and it's not something about I2C/cable/speed/sync/whatever