qmk/qmk_firmware #19743: MacOS Function Key

spencerthayer, 2023-02-03:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Use case: I have a keyboard that has a "microphone" button that returns the key combo `Function + Space` and it's recognized by Better Touch Tools as `Function + Space`.

My QMK keyboard does not have a `Function` key so `Function + Space` is not a viable key combination.

How do I create a Function key with QMK/VIA/VIAL? I understand that `MO()` is the stand in QMK function key but Apple has started mapping more and more things to FN since OS 11, so there is a need to keep the FN key separate from a layer swap. 

Any help would be great. Thank you.

fauxpark, 2023-02-03:
See #2179

spencerthayer, 2023-02-03:
> See #2179

@fauxpark but the Keychron series of QMK keyboards are real apple verified keyboards so legally adding the function as an option would be fine for those keyboards that can use it.

closed 2023-02-03 by fauxpark

spencerthayer, 2023-02-03:
Even if a keyboard company is stealing the code that doesn't make QMK software liable for another company's theft of intellectual property. That's a bit of a weird take.

fauxpark, 2023-02-03:
It's not about code, it's about the USB Vendor ID.
https://www.oshwa.org/2013/11/19/new-faq-on-usb-vendor-id-and-product-id/