qmk/qmk_firmware #19836: I'm compiling and nothing happens and it just ends, right

1219008126, 2023-02-13:
### Issue Description

[image]I have tried other keyboards in qmk and they compile normally, but this one doesn't

elpekenin, 2023-02-13:
That kind of "empty" compilations is likely related to wrong/missing files.
Since the keyboard doesn't exist on the main repo and neither you have a (public) fork of QMK, i cant really help diagnose any further.

1219008126, 2023-02-14:
I used qmk doctor to check and found no problem.I'm not familiar with how to use github, I create a new  forks and upload my files

elpekenin, 2023-02-14:
doctor is not useful here, it simply checks the installation, ie: if you have a git repository cloned and the needed software installed

ForsakenRei, 2023-02-14:
Are you on Windows? If yes you probably want to try `qmk compile -kb citrus/erdnuss65 -km via` and see if it is still the same.

elpekenin, 2023-02-14:
Nop. `qmk compile` is gonna call `make` at the very end....

Issue found right now on their fork, their folder is named `keymap` instead of `keymaps`(note the lack of **s**), works fine if you add that.

By the way, you pushed the commit to your `master` branch, you should use another branch to upload your code @1219008126

elpekenin, 2023-02-14:
For your information, there's an [official QMK Discord server](https://discord.com/invite/Uq7gcHh) where you will(usually) get faster replies