qmk/qmk_firmware #19840: [Bug] LM(layer, mod) setting the wrong mod state for right-mods

sporkus, 2023-02-13:
### Describe the Bug

## description 
When right-mods are used with `LM()`, the mod state is set as the mods_bit.

sending
`LM(0, MOD_RCTL)`
`LM(0, MOD_RSFT)`
`LM(0, MOD_RALT)`
`LM(0, MOD_RGUI)`

I'm getting the following mod state from the console.
[image]

As described in https://github.com/qmk/qmk_firmware/blob/master/docs/feature_advanced_keycodes.md, this flags the right control mod bit for all the keys above. I am confirming this from karabiner event viewer.
[image]

The mod bits that I'm found in `action_code.h`
[image]

## comments
The left mods all work as expected.

The correct behavior seems to be dropping the 5th bit and shift the first 4bits left, and ORing that to the mod state, but I'm new to QMK and don't know where to look further.  I can attempt to create a patch if I get some guidance.

### Keyboard Used

homemade/ handwired

### Operating System

macOS

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /Users/spork/code/qmk_firmware
Ψ Detected macOS 13.1 (Apple Silicon).
Ψ Git branch: layout-dev
Ψ Repo version: 0.11.3
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-09-18 10:01:17 +0000 --  (0e9d558b5)
Ψ - lib/chibios-contrib: 2022-10-03 18:09:41 +0200 --  (bb8356fb)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [X] Karabiner (macOS)

filterpaper, 2023-02-15:
Does #19845 resolve this issue?

sporkus, 2023-02-15:
> Does #19845 resolve this issue?

@filterpaper Yes, it is producing the correct mod state after the fix. I checked both the mod states from the console and OS events. 

@getreuer Thanks for the elegant work!

closed 2023-02-15 by sporkus

getreuer, 2023-02-15:
Oops, I missed that this bug was open. Thanks @filterpaper and @sporkus for closing the loop and verifying the fix!