qmk/qmk_firmware #19843: [Bug] Some RGB keycodes and configuration didn't work on Adafruit RP2040 Macropad

ForsakenRei, 2023-02-14:
### Describe the Bug

- `#define RGB_MATRIX_DEFAULT_VAL 10` seems have no effect on default RGB brightness
- `RGB_HUI`, `RGB_HUD`, `RGB_SAI`, `RGB_SAD`, `RGB_VAI` and `RGB_VAD` won't change anything of RGB

The unstashed/uncommitted changes should be the doc edit.

### Keyboard Used

adafruit/macropad

### Link to product page (if applicable)

https://www.adafruit.com/product/5128

### Operating System

Windows 10 22H2

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: C:/Users/Shigure/qmk_firmware
Ψ Detected Windows 10 (10.0.19045).
Ψ QMK MSYS version: 1.7.2
Ψ Git branch: develop
Ψ Repo version: 0.19.11
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest develop: 2023-02-10 20:12:35 -0800 (34bc896cf9) -- [Keyboard] Fix merge error with fave84 board (#19808)
Ψ - Latest upstream/master: 2023-02-11 09:41:18 +1100 (1504e9ca4c) -- Fix bear65 (#19805)
Ψ - Latest upstream/develop: 2023-02-13 15:51:51 +0100 (e922b46a86) -- Update pico-sdk to 1.5.0 (#19829)
Ψ - Common ancestor with upstream/master: 2023-02-11 09:41:18 +1100 (1504e9ca4c) -- Fix bear65 (#19805)
Ψ - Common ancestor with upstream/develop: 2023-02-10 20:12:35 -0800 (34bc896cf9) -- [Keyboard] Fix merge error with fave84 board (#19808)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-01-03 19:29:26 +0000 --  (0062927e3)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

N/A

drashna, 2023-03-23:
the default val sets it when eeprom is initialized.  So changing it won't change the value unless you're also resetting the eeprom. 

The rest *should* work.

ForsakenRei, 2023-03-23:
> the default val sets it when eeprom is initialized. So changing it won't change the value unless you're also resetting the eeprom.
> 
> The rest _should_ work.

We actually discussed this a while ago on discord, exactly we two lol.

Yes default VAL was set in eeprom, to make a obvious different I set it to `10` and it should be pretty dim, after flashing the firmware I tried `EE_CLR` keycode but the brightness sees to be the same no matter what value I set it to.
Other RGB keycodes are not working properly since there is no change to the RGB color etc.

I want to see if someone can replicate my issue while RP2040 users are focus on another test about waking up PC/Mac recently, and probably not all RP2040 board will have RGB like the adafruit macropad.

drashna, 2023-04-06:
Have my hands on one of these now, and can confirm that it works properly. default val works, keycodes all work.

ForsakenRei, 2023-04-06:
> Have my hands on one of these now, and can confirm that it works properly. default val works, keycodes all work.

Hmmm, would you mind sharing your keymap etc. with me so I can compile a firmware on my side to try?

drashna, 2023-05-22:
Sorry, meant to get back to you sooner:

https://github.com/drashna/qmk_firmware/tree/keymap/drashna/keyboards/adafruit/macropad/keymaps/drashna

closed 2023-09-12 by ForsakenRei