qmk/qmk_firmware #19848: Question

AndroSteph, 2023-02-14:
### Issue Description

Hello,

I have a big question that I can't understand or find a clear detailed answer (besides English is not my language, I have to translate everything with google)

keyboards/keychron/q6/iso_encoder/keymaps
- default ?
- keychron ?
- via ?

Here is my question: where can I find the explanation of why there are three categories "default - keychron via" because I made tests and if I compile with "QMK_MSYS", for my Q6 keyboard, the "default" once put on my keyboard it doesn't work, if I compile the "keychron" once put on my keyboard it's OK, I didn't try with "via"

What is the reason ? 

thank you very much for your comments, I don't like to stay in a situation where I don't understand the "meaning of my problem", while you certainly have an answer :-)

and if I want to modify something, locally on my computer with QMK_MSYS, which section should I use?

THANK YOU!

waffle87, 2023-02-14:
`default` keymap — bare minimum keymap with no additional/custom features
`keychron` keymap — Keychron's "personal" keymap with some extra features (none of which should dictate whether your keyboard works or not). See the `diff` below and feel free to inspect those functions yourself.
`via` keymap — keymap made for use with [VIA Configurator](https://www.caniusevia.com/)
```diff
❯ diff default keychron
diff default/keymap.c keychron/keymap.c
> #include "keychron_common.h"
# ...
> void housekeeping_task_user(void) {
>     housekeeping_task_keychron();
> }
>
> bool process_record_user(uint16_t keycode, keyrecord_t *record) {
>     if(!process_record_keychron(keycode, record)) {
>         return false;
>     }
>
>     return true;
> }
Only in keychron: rules.mk
```
I don't see any reason the `default` keymap wouldn't work for you, but the `keychron` one does. In general, you should use `qmk new-keymap` to create your own personal, but that makes a copy of the `default` keymap. If that's not working for you, perhaps it's best to just make a copy of the `keychron` keymap. I'd also suggest joining the QMK Discord for further (conversational) help.

AndroSteph, 2023-02-15:
Hello waffle87,

Thank you very much for your clear explanations!

I managed to compile the "keychron" version having even changed the lighting of the "capslock" key to red (instead of white), I am very happy.

Thanks again for the time you spent to answer me :-)

Steph

AndroSteph, 2023-02-15:
I also add as information, that the "keychron" version, at least for me, works with "VIA". So I am totally happy.

In addition you have added / changed the function (original of Keychron) of the keys "fn+J+Z" which now do not work with the key "ESC + USB cable connection", it's great because now it is not necessary to remove the "space" key to press the physical button "reset" of the keyboard to be able to put a new firmware. **THANK YOU !!!!**

closed 2023-02-15 by AndroSteph