qmk/qmk_firmware #19855: [Bug] compiling configurator json does not include custom rules.mk or source.c

dustinlagoy, 2023-02-15:
### Describe the Bug

I have been build firmware following https://docs.qmk.fm/#/newbs_building_firmware_workflow so I can keep my keymap in revision control without forking and it works great with my pure json keymap! However, when I try to add anything into `sources.c` it doesn't seem to end up in the compiled keymap. In fact, `sources.c` doesn't seem to be included at all in the compilation, even though it is added via `rules.mk`. This happens on my local machine when running `qmk compile path_to_json_file` or via the recommended github workflow.

Looking at the python sources the function `compile_configurator_json` (https://github.com/qmk/qmk_firmware/blob/master/lib/python/qmk/commands.py#L110) doesn't seem to use any file but the passed json.

Is this by design? If so then the documentation linked above is confusing to me. Generally I'm happy to compile with pure c keymap files but the ability to compile out-of-tree is appealing.

Sorry if I'm missing something obvious :)

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

drashna, 2023-02-17:
To clarify, this is the keymap json exported from the configurator, correct? 

And do you mean it's not including the keymap files or the keyboard files?

dustinlagoy, 2023-02-21:
Thanks for reaching out! I was following along with https://docs.qmk.fm/#/newbs_building_firmware_workflow so I have a repository containing the following files:

```
atreus.json  config.h  rules.mk  source.c
```

The json file is following the configurator conventions but I wrote it by hand. I made a link to this folder at `qmk_firmware/users/dustinlagoy`.

When I run `qmk compile users/dustinlagoy/atreus.json` I successfully build a firmware that includes everything in `atreus.json`. The issue is that neither `rules.mk` or `source.c` are included.

filterpaper, 2023-02-22:
Can you share the content of `atreus.json`? Does it contain a keymap value with your user name? e.g.:
```c
"keymap": "dustinlagoy"
```

dustinlagoy, 2023-02-23:
Wow thanks @filterpaper! That fixed my issue. The guide I was following clearly said to use my username in the keymap field but for whatever reason I did not so sorry for needlessly opening this issue. I tried to look through the source for how those files might be included but didn't do a thorough enough job. Now I see in https://github.com/qmk/qmk_firmware/blob/master/builddefs/build_keyboard.mk#L356 how the keymap name is used to find the source files!

Thanks again :)

closed 2023-02-23 by dustinlagoy