qmk/qmk_firmware #19871: [Bug] Teensy 2.0 handwired pad key stuck after Windows resume from sleep

skagon, 2023-02-17:
### Describe the Bug

I have a number of custom (self-made) keypads, all with my own QMK keyboard code and layouts. There is a problem when Windows resumes from sleep. After pressing a key on one of the keypads, which causes the computer to wake up, the key that I pressed on the keypad will remain pressed until I press any key (the same or other) on that pad. For instance, if I press [left arrow], I can't input the Windows PIN because the cursor is always moving to the left. If it's some other key, I either get spammed by keypresses, or if it's some modifier key, I'll log in but then it's pressed all the time, until I press any key on that particular keyboard.
I've seen it happen on three of my custom pads.
I'm on the latest, but it happened even before updating the master.

### Keyboard Used

Teensy hand-wired (multiple)

### Operating System

Windows 10 22H2

### qmk doctor Output

```text
[will add later tonight]
```

### Is AutoHotKey / Karabiner installed

- [X] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

Perhaps related to https://github.com/qmk/qmk_firmware/issues/19600 and/or https://github.com/qmk/qmk_firmware/issues/16969 ?

dansity, 2023-05-06:
I have the same issue with the same board, the other merged solutions did not resolve it for this board.

skagon, 2023-08-02:
> does adding `#define USB_SUSPEND_WAKEUP_DELAY 200` to your config.h help?

Ok, update: it does not work. At least, not consistently. My previous comment was made after putting the PC to sleep myself, from the start menu. Then re-awakened it within a few seconds of it turning off a.k.a. sleep). Then everything seemed normal. But after the PC goes to sleep after a period of inactivity, same things happen, which means keys still get stuck apparently randomly. I've even had a keypad (with an Enter key) get stuck after I pressed the Enter key, but it did NOT wake up the PC. I had to press a button on a different keypad/board to wake up the PC, but the Enter key on the first keypad was still stuck! For the record, the keypads are all connected through a powered USB hub, so they don't need PC power to turn on.
So… what now?